Ysurac/openmptcprouter #3290: Vpn tunnel down while tunnel is not down

Brazzo978, 2024-04-17:
Bring back Timeout of more than 10s in OMR-Tracker like it was on 0.59 and lower 

## Current Behavior
when setting OMR-Tracker for omrvpn and other interface now you can select a ping timeout from 1 second to 10 second , in my case the connection sometimes need >10s of time to complete the ping command but the link is still up , infact the command ping 10.255.255.1 show 0% packet loss in the omr terminal , 

## Possible Solution
bringing back the text field like on 0.59.x

## Specifications

  - OpenMPTCProuter version: 0.60.1
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider: Own
  - OpenMPTCProuter platform: x86
  - Country: italy

Ysurac, 2024-04-17:
I will add the possibility to add any custom timeout.
You really have latency more than 10s and something is working ? It's only the VPN or you have bad connections ?

Brazzo978, 2024-04-17:
Yes i have starlink + 2 lte connection , when starlink is set as primary glorytun used to do more than 10000ms ping , while downloading from my local nas from the outside , now with openvpn ping is much lower in 0.60 but also is bandwidth , before with glorytun tcp i got around 30MB/s now i cant get 20MB/s stable and i am running both version at the same time also some friend of me with the same setup confirmed they used 12/20s of timeout and tuned glorytun timeout also , its slow but packet loss was near 0%

closed 2024-04-19 by Brazzo978