Ysurac/openmptcprouter #3294: eth0 carry on other usb device,not the board at Recent version

fareign, 2024-04-17:
if I have a usb networkcard on the router,eth0 allways gone to the usb networkcard not the board.
I'm not sure it is relate to the linux kernel. or openwrt version

## Specifications

  - OpenMPTCProuter version: 0.60 6.1
  - OpenMPTCProuter VPS version: 101029
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: /x86_64
  - Country:

Ysurac, 2024-04-17:
This is by detection order.
You can make an interface to always keep same name by unchecking box "Disable interfaces auto rename" in System->OpenMTPCProuter, "Advanced settings" tab. This can sometimes help.

fdyf520, 2024-04-18:
按照上面的设置之后并没有生效，依旧是原先的效果

extozi, 2024-04-19:
I also have the same problem, it came after openmptcprouter-v0.60beta1-6.1-r0+23789
I still couldn’t get eth0 to remain on the system rj45 after a reboot. If you unplug 2 USB cards before starting, the system port on the mini-pc board is assigned eth0.
so far I have rolled back where there is no such problem

closed 2024-07-23 by github-actions[bot]