Ysurac/openmptcprouter #3297: OMR ByPASS not working

Majaxx80, 2024-04-18:
## Expected Behavior
Pass the flow through wan1 or wan2
## Current Behavior
After configuring OMRBYPASS the flow always goes via the vps

[image]

`root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'lpsn.net' 'lpsnmedia.net' 'francetelecom.fr'
omr-bypass.eth0=interface
omr-bypass.eth0.id='9999'
omr-bypass.wan1=interface
omr-bypass.wan1.id='7'
omr-bypass.wan2=interface
omr-bypass.wan2.id='8'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.global=global
omr-bypass.global.vpn_ipv4_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.global.vpn_ipv6_md5='68b329da9893e34099c7d8ad5cb9c940'
omr-bypass.@lan_ip[0]=lan_ip
omr-bypass.@lan_ip[0].ip='192.168.100.235'
omr-bypass.@lan_ip[0].interface='wan1'
`
## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 0.10.30
  - OpenMPTCProuter VPS provider: PulseHeberg
  - OpenMPTCProuter platform: Dell
  - Country: FR

Ysurac, 2024-04-19:
ByPass all traffic from an IP is now fixed. This will be in next snapshot (it's compiling) so in next release.

winkmichael, 2024-04-25:
This will be an RC3?

> ByPass all traffic from an IP is now fixed. This will be in next snapshot (it's compiling) so in next release.

Ysurac, 2024-04-25:
This will be in v0.60.1

closed 2024-07-29 by github-actions[bot]