Ysurac/openmptcprouter #3298: Impossible to setup MPTCP over VPN.

vempire-ghost, 2024-04-19:
## Expected Behavior
Enabling MPTCP over VPN works

## Current Behavior
The interface ovpnwan1 show IP but with Gatway DOWN.

## Steps to Reproduce the Problem
Select the MPTCP over VPN in the wizard for WAN1.

## Context (Environment)
I'm running some tests that require the use of MPTCP over VPN, but I can't seem to get it to work.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-snapshot-5.4 r0+16862-1
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1030
  - OpenMPTCProuter VPS provider: Oracle
  - OpenMPTCProuter platform: x86_64
  - Country: 

Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[17653]: /usr/libexec/openvpn-hotplug down wan1 tun10 1500 0 10.255.250.6 10.255.250.5 init
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[17653]: SIGTERM[hard,] received, process exiting
Apr 19 01:19:35 OpenMPTCProuter daemon.warn openvpn(wan1)[416]: DEPRECATED OPTION: --cipher set to 'AES-256-CBC' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
Apr 19 01:19:35 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to xxx.138.151.xxx dev tun10
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: OpenVPN 2.6.10 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: library versions: OpenSSL 1.1.1t  7 Feb 2023, LZO 2.10
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: DCO version: N/A
Apr 19 01:19:35 OpenMPTCProuter daemon.warn openvpn(wan1)[416]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Apr 19 01:19:35 OpenMPTCProuter daemon.warn openvpn(wan1)[416]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: TCP/UDP: Preserving recently used remote address: [AF_INET]xxx.138.151.xxx:65301
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: UDPv4 link local (bound): [AF_INET]192.168.0.50:65301
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: UDPv4 link remote: [AF_INET]xxx.138.151.xxx:65301
Apr 19 01:19:35 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to xxx.138.151.xxx dev tun10
Apr 19 01:19:35 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: [server] Peer Connection Initiated with [AF_INET]xxx.138.151.xxx:65301
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: DCO device tun10 opened
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: net_iface_mtu_set: mtu 1500 for tun10
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: net_iface_up: set tun10 up
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: net_addr_ptp_v4_add: 10.255.250.6 peer 10.255.250.5 dev tun10
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: /usr/libexec/openvpn-hotplug up wan1 tun10 1500 0 10.255.250.6 10.255.250.5 init
Apr 19 01:19:52 OpenMPTCProuter daemon.notice netifd: Interface 'ovpnwan1' is enabled
Apr 19 01:19:52 OpenMPTCProuter daemon.notice netifd: Network device 'tun10' link is up
Apr 19 01:19:52 OpenMPTCProuter daemon.notice netifd: Interface 'ovpnwan1' has link connectivity 
Apr 19 01:19:52 OpenMPTCProuter daemon.notice netifd: Interface 'ovpnwan1' is setting up now
Apr 19 01:19:52 OpenMPTCProuter daemon.notice netifd: Interface 'ovpnwan1' is now up
Apr 19 01:19:52 OpenMPTCProuter daemon.notice openvpn(wan1)[416]: Initialization Sequence Completed
Apr 19 01:19:52 OpenMPTCProuter daemon.info ModemManager[8129]: hotplug: add network interface tun10: event processed
Apr 19 01:19:52 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of ovpnwan1 (tun10)
Apr 19 01:19:52 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 19 01:20:01 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Apr 19 01:20:02 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
Apr 19 01:20:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Apr 19 01:20:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

[image]

Ysurac, 2024-04-19:
Why do you want to use OpenVPN for MPTCP over VPN ?
It was not tested since a long time, so possible... I will check.

vempire-ghost, 2024-04-19:
I also tried using WireGuard, but I encountered the same issue. The WireGuard screen on the OMR reports a handshake problem with the VPS.

I'm trying to explore options to increase OMR's resilience for situations where there's an interruption in the internet connection of the master WAN. As I mentioned some time ago, omr-tracker takes quite a while to remove a failed route. I want to see how it behaves when the master WAN is on a VPN and there's a failure, whether the reconnection is faster or not.

Since I'm using OMR to increase connection resilience, in games that constantly establish new connections, being without internet for several seconds can mean a failure in these new connections and a disconnect.

vempire-ghost, 2024-04-19:
> I working MPTCP over VPN issues.
> 
> This will not help you in any other cases than MPTCP is blocked on a connection. There is lot of options in OMR-Tracker now that can help to switch faster if needed.

I understand, but I like to explore all possible situations. I've achieved many improvements with fine-tuning configurations in OMR. Today, apart from this small window, OMR gives me the best connection resilience I could imagine :)

On another note, I know it's not the scope of this topic, but I wanted to ask: in version 0.59, it was possible to use the configuration `uci set omr-tracker.wan1=interface` `uci set omr-tracker.wan1.script_alert_up=` to define some command in the ifup and ifdown alert, but starting from 0.60, this command apparently no longer works. When applied via SSH, it doesn't save in OMR anymore. Has this command been deprecated or do I need to make some changes for it to work again? It was very useful for me to configure some custom parameters in these events.

closed 2024-07-23 by github-actions[bot]

abadsah, 2024-08-01:
Hi,

im getting the exact same problem, any solution for MPTCP over vpn?

Regards,

Ysurac, 2024-08-01:
MPTCP over VPN Wireguard is working in snapshot image.
Make sure you really need it, it's only a dirty workaround.

abadsah, 2024-08-01:
yeah i really need it i have 4 dsl connections and all blocked multipath tcp 

can you please send me links for the snapshot versions 

im using openmtcprouter image on my old laptop

Regards,