Ysurac/openmptcprouter #3300: OpenVpn Bonding (udp) is not working

Brazzo978, 2024-04-19:
## Expected Behavior
enabling openvpn udp as omrvp should create 3 udp openvp istance and aggregate using openvpn 
## Current Behavior
system loses connection to vps and to the internet , everything stop working 

## Steps to Reproduce the Problem

  1. enable openvpn bonding with any proxy
  2. everything stop working 

## Context (Environment)
https://pastebin.com/2c8b86rC

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 1030
  - OpenMPTCProuter VPS provider: own
  - OpenMPTCProuter platform: x86_64
  - Country: italy

Brazzo978, 2024-04-19:
MLVPN is unable to connect to vps part too :  if that can help Apr 19 12:07:10 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Apr 19 12:07:10 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: sh: missing ]
Apr 19 12:07:12 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: Device "mlvpn0" does not exist.
Apr 19 12:07:12 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: Cannot find device "mlvpn0"
Apr 19 12:07:12 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: Cannot find device "mlvpn0"
Apr 19 12:07:12 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: Cannot find device "mlvpn0"

Ysurac, 2024-04-19:
I'm not able to reproduce the issue.
OpenVPN bonding never worked correctly, seems top be working but loosing lot of packets.
I don't have issues with mlvpn, do you have more log about mlvpn ?

Brazzo978, 2024-04-19:
hi i was testing and for test purpose i just switched in sqm advanced setting i put 0 on all red box in the sqm configuration , then i put fq_codel as algorithm and simplest.qos as queue setup script , now mlvpn started working , glorytun ping dropped 10s and i am able again to use full bandwidth in upload , is possible that there was a bug with the scheduler / the red checkboxes that made sqm work in a weird way ?

sp2003, 2024-04-20:
@Ysurac  The tunnel has autorate enabled by default. Is this regular?

Ysurac, 2024-04-20:
@sp2003 should be disabled in latest snapshot

Brazzo978, 2024-04-22:
@Ysurac  when and where will be available the latest snapshot ?

Ysurac, 2024-04-22:
It's always available in https://snapshots.openmptcprouter.com/

Brazzo978, 2024-04-25:
@Ysurac  what about the  vps part , does the normal script work for 0.61 snap ?

Ysurac, 2024-04-25:
Yes it will work. Snapshot script also fix some issues.

closed 2024-07-29 by github-actions[bot]