Ysurac/openmptcprouter #3301: The ability to define which WAN the proxy will use to initiate the connection?

vempire-ghost, 2024-04-19:
Currently, OMR routes all proxy traffic through the master WAN. Is it possible to configure the proxy to use another WAN to initiate connections instead of the master? This would allow keeping the master on one WAN, but having the proxy use another WAN to establish initial connections.

Ysurac, 2024-04-19:
I don't understand what you mean. Master define interface used for initial connection, it's the default gateway, so why using another connection and what would be master in this case ?

vempire-ghost, 2024-04-19:
I just want to direct the proxy connection to a WAN without having to define it as the default gateway. Perhaps using some firewall rule for that?

aladdinshammr, 2024-04-22:
@vempire-ghost 
What I could not understand is "What is the use of backup interface?"
It just stays idle there, and when the master connection goes down backup connection keeps sleeping.

Ysurac, 2024-04-22:
I don't think it's related to the subject, but backup interface is (or at least should be) used when master and interface set to "on" are down.

closed 2024-07-27 by github-actions[bot]