Ysurac/openmptcprouter #3306: Is it possible to run OMR on mikrotik router board?

aladdinshammr, 2024-04-22:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

I found some articles that talk about installing openwrt on a mikrotik router board RB1100AH and another types.
I wonder if it is possible to run OMR on RB1100AHx4 or RB1100AHx2  , and if it possible which IMG should I use.
most of middle east counties use this product, therefore it will be a good idea to implement these types of routers

**Describe why you are asking for this support?**

This router comes with many ports, instead of using one for all connections, which I think it increases the latency.

**Additional context**

https://wiki.mikrotik.com/wiki/Manual:Metarouter
https://tomas-savenas.medium.com/openwrt-on-mikrotik-as-a-metarouter-2b0bed9d62f9
https://mikrotik.com/product/rb1100ahx4#fndtn-specifications

Ysurac, 2024-04-22:
Not really a cheap product...
Router OS 7 is not compatible with OpenMPTCProuter, I think this product is still using Router OS 6, so should be possible to compile OpenMPTCProuter for it: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
But I don't think the CPU can provide a fast connection...

aladdinshammr, 2024-04-22:
I use it for load balance.
could you recommend a good router?

closed 2024-07-27 by github-actions[bot]