Ysurac/openmptcprouter #3307: VPN tunnel down  VPS v0.1030 6.1.0-18-amd64, OMPTCPR v0.60-6.1

DCNULL, 2024-04-23:
[image]

When using v0.59.1 and OpenMPTCProuter VPS 0.1028, everything worked. I installed (did not update) VPS OpenMPTCProuter VPS 0.1030 and v0.60-6.1 and got VPN tunnel down. The provider is the same, the OpenMPTCProuter settings are by default.

system log
https://upaste.me/42b8573991034a18a

## Specifications

  - OpenMPTCProuter version: v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-04-23:
In the log it's seems that the VPN is up.
What is the result, via SSH on the router, of ```ip r``` (remove VPS IP before pasting) ?

Ysurac, 2024-04-23:
ok, so seems to be only wrong display, maybe because 10.255.x range is also used here. I will check.

DCNULL, 2024-04-23:
Changed 10.200.xx/8
root@OpenMPTCProuter:~# ip r
default via 10.255.252.1 dev tun0
default metric 1
        nexthop via 10.200.1.254 dev wan1 weight 100
        nexthop via 10.200.2.254 dev wan2 weight 1
default via 10.200.1.254 dev wan1 metric 7
default via 10.200.2.254 dev wan2 metric 8
default via 10.255.252.1 dev tun0 metric 1500
10.0.0.0/8 dev wan1 scope link metric 7
10.0.0.0/8 dev wan2 scope link metric 8
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.3
10.255.252.0/24 dev tun0 scope link metric 1500
10.255.252.1 via 10.255.252.1 dev eth0
127.0.0.0/8 dev lo proto static scope link metric 5
141.8.x.x metric 1
        nexthop via 10.200.1.254 dev wan1 weight 100
        nexthop via 10.200.2.254 dev wan2 weight 1

It's all the same.

[image]

Ysurac, 2024-04-23:
I think this is the problem: "10.255.252.1 via 10.255.252.1 dev eth0" I need to check how you can have this dirty route...

DCNULL, 2024-04-23:
root@OpenMPTCProuter:~# ip -4 r list dev tun0
default via 10.255.252.1
default via 10.255.252.1 metric 1500
10.255.252.0/24 proto kernel scope link src 10.255.252.2
10.255.252.0/24 scope link metric 1500

Ysurac, 2024-04-23:
This will be fixed in next release. Should be only a display issue.
I still need to find where this route is coming from...

closed 2024-04-23 by DCNULL