Ysurac/openmptcprouter #3314: Traffic transmitted on the disabled WAN interface.

yue2388253, 2024-04-25:
## Expected Behavior

The traffic MUST NOT be transmitted on the WAN interface that has been already disabled, as shown in the figure.
[image]

## Current Behavior

Even though the WAN interface is disabled, the traffic is still transmitted via this interface.

## Possible Solution

## Steps to Reproduce the Problem

  1. Say I have two WAN interfaces, WAN1 and WAN2.
  2. Disable WAN1 in the OpenMPTCPRouter settings.
  3. Set WAN2 as the master inferface.
  4. Physically disconnect WAN2 so that the Internet is not available on WAN2, as shown in the figure.
  5. Now in theory the router is supposed to disconnect from the Internet. However, it can still access the Internet via the disabled WAN1, even though the WAN1 is not shown in the `Network overview` figure.

[image]

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  0.59.1-5.4 
  - OpenMPTCProuter VPS version: 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Alibaba cloud
  - OpenMPTCProuter platform:  x86
  - Country:  China

Ysurac, 2024-04-25:
The interface in disabled mode, is disabled for MPTCP only. So the interface will not be used for aggregation. But the interface is still available on the router.
The interface need to be disabled via Network->Interfaces menu.

yue2388253, 2024-04-25:
Thank you for your quick reply.

Would it be better to enhance the UI to avoid ambiguity? Maybe something like the following figure, where the disabled interface is still shown yet it shows "Multipath disabled" in red.

[image]

closed 2024-07-29 by github-actions[bot]