Ysurac/openmptcprouter #3321: Is it possible to compile OMR using a forked version of MPTCP 0.96 instead of the official one in kernel 5.4?

vempire-ghost, 2024-04-26:
Would it be possible to compile OMR on version 5.4 using a forked version of MPTCP 0.96 where I made some changes? If so, could you guide me on how to do it?
By the way, do you use the final version 0.96 or one with some of your modifications?

Ysurac, 2024-04-26:
It's 0.96. There is modification only on x86 to allow more subflows.
MPTCP is added as a simple patch, remove it and add another one https://github.com/Ysurac/openmptcprouter/blob/develop/5.4/target/linux/generic/hack-5.4/690-mptcp_v0.96.patch

vempire-ghost, 2024-04-26:
So I just need to compile the patch and replace it with this file you linked after cloning your repository and before compiling?

vempire-ghost, 2024-04-26:
Can I just edit your patch by adding the modifications to it before compiling?

closed 2024-07-31 by github-actions[bot]