Ysurac/openmptcprouter #3322: Config restore from 5.4 to 6.1 extreme CPU Usage

darkmaniac7, 2024-04-26:
Recently created a new VM and restored the old config from the 0.59 - 5.4 OS.  New VM was 0.60 - 6.1 OS. it used to Typically run under 700mhz usage. Now runs over 11ghz most of the time.

## Expected Behavior
CPU usage while idle be less than 1ghz usage 

## Current Behavior
Idle usage between 9.7Ghz-11.8Ghz. 12Ghz on speedtest.
htop shows almost no cpu usage by processes
[image]

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version:  0.1030 6.1.0-20-amd64
  - OpenMPTCProuter VPS provider: Kamatera
  - OpenMPTCProuter platform: vmware EESXi 7.0.3, VPS OS - Debian 12

Kernel Log:
[https://pastebin.com/StDBNtnE](url)

Ysurac, 2024-04-26:
If your VM allow it (else the option is not displayed), you can try to change Scaling governor in System->OpenMPTCProuter, "Advanced settings" tab.

darkmaniac7, 2024-04-26:
> If your VM allow it (else the option is not displayed), you can try to change Scaling governor in System->OpenMPTCProuter, "Advanced settings" tab.

Unfortunately it doesn't appear that it does, can I make any changes to it elsewhere or any other logs that may indicate what's taking up so much CPU? Performance is fine, it just uses a lot more CPU now.

Ysurac, 2024-04-27:
If it doesn't appear than you can't change it.
I don't have this problem on x86_64 but I'm using 6.6 snapshot now.
What do you have in Status->System log ?

darkmaniac7, 2024-04-27:
This is what I have, most of the identifying info is changed. Let me know if you need the original.

https://pastebin.com/Nb2fAnTu

Ysurac, 2024-04-27:
Seems that VPN Glorytun doesn't work correctly. Try another one in System->OpenMPTCProuter, Wizard tab, advanced settings checkbox. Default is now "OpenVPN TCP".

darkmaniac7, 2024-04-27:
> Seems that VPN Glorytun doesn't work correctly. Try another one in System->OpenMPTCProuter, Wizard tab, advanced settings checkbox. Default is now "OpenVPN TCP".

That fixed that issue in the System Log, thanks! 

Same issue regarding CPU usage however, which is strange since no process appears to be using it more than 2% CPU.

Wlanfr3ak, 2024-05-02:
I have the same CPU bug (started a few days after upgrading). I now fresh installed router and dont use backup file and reconfigured manually. Will test but the first steps are looking better.

darkmaniac7, 2024-05-07:
> I have the same CPU bug (started a few days after upgrading). I now fresh installed router and dont use backup file and reconfigured manually. Will test but the first steps are looking better.

I saw the same, but when setting a lot of the settings the same I encountered VPN issues (IPSEC VPN with another Location), Teamviewer and port forwarding not working, etc. So I just stuck with the backup config. When doing a fresh install and rebuilding the config it does lower CPU Usage to typical levels, but for me is unusable even if I disable shorewall on the VPS Port Forwards, VPN, etc. didn't work.

Not really an issue but when using the 5.4 to 6.1 config the WAN Sources show up as not working but internet is working just fine. Not sure if there is a way to fix that either. 

Ultimately the server is an i9-11900F in a mini 1u server so it can handle that load. But I'd prefer it didn't.

[image]

Ysurac, 2024-05-07:
You can try 6.6 snapshot (it will work with VPS on 6.1).

Ysurac, 2024-05-07:
Do you have high CPU usage on idle or only when using it ?

darkmaniac7, 2024-05-07:
> Do you have high CPU usage on idle or only when using it ?

Both, When Idle and using it doesn't seem to change much. 11-14Ghz

[image]

darkmaniac7, 2024-05-15:
Just wanted to update. I tried loading the config from 5.4 to 6.6 but it resulted in the same very High CPU usage. Eventually I just rebuilt from scratch using the 6.6 Snapshot and all seems well now, port forwards are working and CPU usage is back down to 4-750mhz.

hhaniel, 2024-06-28:
I found the problem to be down to the omr-tracker - I now do a ping test every 10 seconds or so and that has reduced the cpu usage of the VM quite drastically - I am also hoping that reducing the agressiveness of the OMR tracker may stop my interfaces going up and down like a yo yo.

closed 2024-10-01 by github-actions[bot]