Ysurac/openmptcprouter #3327: OpenMPTCProuter Snapshot

Ysurac, 2024-04-30:
**Common:**
* Update Proxy and VPNs
* MPTCP over VPN fixes
* XRay/V2Ray port forward fix

**Kernel 6.1/6.6:**
* Fixes on OMR-ByPass
* N100 support

**Kernel 6.6:**
* Add MPTCP Schedulers via BPF

To download images: https://snapshots.openmptcprouter.com/

_For VPS part:_
For the develop server script (same script for Debian 9/10/11/12 and Ubuntu):
```wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.1" sh```

For kernel 6.6 You need to also use router part with kernel 6.6:
```wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh```

jle1511, 2024-05-01:
we tested both updated to this snapshot (router and vps)

We test with proxy V2RAY Vless and Xray Vless 
after reload reboot:
curl ifconfig.me is working.

So if we change proxy, do we need to reboot for omrbypass to reload correct rule, or simply just save and apply

hle5128, 2024-05-01:
@Ysurac Does omrbypass over VPN work in this snapshot? 
Thanks

iazxq, 2024-05-06:
@Ysurac After using this snapshot and rebooting the system, I found that the network card was not recognized, and eth0 could not be found. My network card model is: Realtek 8111E.

esquilo444, 2024-05-08:
hello,
wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh doesn't work for me .
i don't undersdand why ...
[image]

esquilo444, 2024-05-08:
thank you in advance for your help

esquilo444, 2024-05-08:
it work perfectly with the 6.1 version

vinceducat, 2024-05-09:
@esquilo444 i have the same problem with vps in Hetzner..  (the original kernel of the vps is 6.1 perhaps a problem, @Ysurac do we update first the original debian 12 vps to kernel 6.6 and after doing install off omr )

vinceducat, 2024-05-09:
@esquilo444  perhaps i will be better ton rename the title of the post and add something with "dl.xanmod.org" to help other users ;-)

Ysurac, 2024-05-10:
@vinceducat Update to debian 12 is made by the script.
It seems that xanmod block Hetzner IP range at least, I will make a mirror.

vinceducat, 2024-05-10:
@Ysurac thanks ;-) 
wait for the new script so ;-)

Ysurac, 2024-05-13:
Script updated, it now use latest Debian kernel 6.6

vinceducat, 2024-05-13:
@Ysurac  thanks test server is install on Hetzner vm in debian 12.. with 6.6 kernel.

my first goal was ton use only ipv6 beetwen OMR router and OMR vps... how can i achieve this .. ?

i have ipv6 adresse on the server 
i have one Frebox pop... how to setup an ipv6 only connexion ... the template seem to need ipv4 and ipv6

esquilo444, 2024-05-13:
Thanks a lot....

Alison110, 2024-05-25:
Good morning, Is it possible to use (OpenWRT) geosite/geoip packages with v2ray/xray ? (to bypass all CN or Fr or Ru or …sites. Something similar to v2raya or passwall bypass)

tokon2000, 2024-06-07:
Kernel 6.6 script broken again on Hetzner VPS?
Fails again at the xanmod part.

Ysurac, 2024-06-09:
Yes, I forgot this part... I migrated to Debian kernel 6.6 but it was not working on some provider, so I migrated again to xanmod. I will create a mirror, really this time I put it on todo list.

Ysurac, 2024-06-09:
I've updated the VPS snapshot install script with local mirror, but not yet tested.

tokon2000, 2024-06-09:
> I've updated the VPS snapshot install script with local mirror, but not yet tested.

Thanks, did a quick test. Installation was possible, server is running.

But this is also reproducible, so no difference between 6.1 and 6.6:
https://github.com/Ysurac/openmptcprouter/issues/3397

Brazzo978, 2024-07-23:
does the last snapshot have wan tiers ?

darkman1983, 2024-07-27:
On [openmptcprouter v0.61-snapshot-6.6 r0+26695-820823198d] i'm not able to use:
Glorytun UDP
Openvpn Bonding

Glorytun UDP status on VPS is OK.
Glorytun UDP on Router says active.

But Status Page says not VPN-Connection...
Can't seem to find any cause...

darkman1983, 2024-07-31:
With Snapshot-061 Glorytun-UDP is working again! However the overall latency got more worse than with 6.6 Kernel :-D

Alison110, 2024-08-01:
Sometime I manually restart glorytun from startup and after 2 3 minute all status are OK,
this last snapshot has alot of bugs, after install  It can not find server script and it can not downgrade or upgrade! You must first reset it To default settings then start to downgrade, best snapshot 6.6 was the one before 2 latest release .(Third).You should try that .
If ysurac add sing-box with shadow tls and revers proxy with grpc to this snapshot Ping and latency gonna be Very low and stable

darkman1983, 2024-08-02:
Currently on 0.61 snapshot with 6.10 Kernel you couldn't reset anything if admin script isn't reachable. 
If i use a cribbing to kill omr-admin.py every x minutes, it works again for 1 minute and then stops to work again. 

Is there any log file for omr-admin.py on VPS?

Ysurac, 2024-08-02:
I don't understand what you mean about reset.
```journalctl -u omr-admin``` to check if you have issue.
I'm making some update to the script every day for now, another one will be out today.

darkman1983, 2024-08-05:
With latest Snapshot-Updates admin script is stable again. glorytun-udp is working :-)

Ysurac, 2024-08-13:
There is no changes on Glorytun-UDP in OpenMPTCProuter since a long time...
Glorytun-UDP is not really verbose.
You can use the ```glorytun-udp``` command via SSH on the router to check status, for example ```glorytun-udp path```.

darkman1983, 2024-08-15:
I already did and there's no error tho. But Router Part tells the VPN is down. Always around 20-30 min of runtime.
Maybe a memory leak or something in that direction?

dersch81, 2024-08-26:
I have a Brand New N100 embedded device here which falls in a boot loop with stable image. The latest OpenWRT is working with the device. So I want to try the snapshot but when i dd the image to the nvme then I don't have a boot partition. Any advice for me how to use the snapshot with a frech install?