Ysurac/openmptcprouter #3328: Network dies when 100+ users are using the bonding

QCWBradleyGebhardt, 2024-05-01:
## Expected Behavior

The network stays stable with 100+ users

## Current Behavior

The network is stable and usable with 5 users watching videos and some browsing. As soon as we flick the other 100+ users on to it the network stops responding. We were unable to even load the router page when connected to it via ethernet. Web traffic dies as well, with pings somehow still being stable

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: A proxmox VM hosted on Xneelo
    - CPU: 8 core  
    - RAM: 16GB
  - OpenMPTCProuter platform: x86_64
    - CPU: Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
    - RAM: 10GB

## Bonded links

  - 200Mbps fibre
  - Rain 5G ~120Mbps (varying speeds)
  - Rain 5G ~120Mbps (varying speeds)

Watching the htop for the router and the VPS, the router reached a max of around 30% across all 4 cores and the VPS was below 10%. Using nload we could also see that there was traffic flowing (~50Mbps - ~200Mbps+)

## Router Config

[Pastebin Link](https://pastebin.com/hbH2Cvg6)

Ysurac, 2024-05-02:
Can you try to update to latest 0.60 release with 6.1 kernel ?

QCWBradleyGebhardt, 2024-05-06:
We've done some of the small scale testing today. After the updates we now notice that the traffic is primarily flowing over whichever link is set as the master link. With only a few KB going over the other links. We left the settings unchanged (besides changing the master links), only running the new tests after the update. 

Besides that it's stable so far. We're waiting to fix this issue before doing the large-scale testing.

[Settings after the update](https://pastebin.com/kXHD5F4n)

QCWBradleyGebhardt, 2024-06-10:
We've now rolled back to the original versions listed in the issue and traffic is pushing over all the links again

closed 2024-09-13 by github-actions[bot]