Ysurac/openmptcprouter #3330: Bonding is not working when IPV6 ( Only ) is used.

iamdivyanshverma, 2024-05-03:
Hi @Ysurac Hope you are doing well.

## Current Behavior

I have 2 WAN connections both IPV6 only, dialed with QMI . Both ping individually but no Internet on LAN. On the status page it says gateway down. 

 ############### wan1 
 root@OpenMPTCProuter:~# ping6 -I wwan0 2001:4860:4860::8888
PING 2001:4860:4860::8888 (2001:4860:4860::8888): 56 data bytes
64 bytes from 2001:4860:4860::8888: seq=0 ttl=117 time=60.517 ms
64 bytes from 2001:4860:4860::8888: seq=1 ttl=117 time=78.238 ms
64 bytes from 2001:4860:4860::8888: seq=2 ttl=117 time=60.422 ms
^C
--- 2001:4860:4860::8888 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 60.422/66.392/78.238 ms

################ wan2
root@OpenMPTCProuter:~# ping6 -I wwan1 2001:4860:4860::8888
PING 2001:4860:4860::8888 (2001:4860:4860::8888): 56 data bytes
64 bytes from 2001:4860:4860::8888: seq=0 ttl=117 time=56.904 ms
64 bytes from 2001:4860:4860::8888: seq=1 ttl=117 time=44.260 ms
64 bytes from 2001:4860:4860::8888: seq=2 ttl=117 time=54.596 ms
^C
--- 2001:4860:4860::8888 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss

## Network file:
config interface 'wan1'
	option proto 'qmi'
	option device '/dev/cdc-wdm0'
	option apn 'JIONET'
	option auth 'none'
	option pdptype 'ipv6'
	option multipath 'master'
	option addlatency '0'
	option metric '8'
	option ipv6 '1'
	option defaultroute '0'
	option peerdns '0'

config interface 'wan2'
	option proto 'qmi'
	option device '/dev/cdc-wdm1'
	option apn 'airtelgprs.com'
	option auth 'none'
	option pdptype 'ipv6'
	option multipath 'on'
	option addlatency '0'
	option metric '9'
	option ipv6 '1'
	option defaultroute '0'
	option peerdns '0'

## DHCP file
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra_slaac '1'
	option force '1'
	option dhcpv6 'server'
	option ra 'server'
	option ra_default '1'
	option ra_preference 'high'
	option ra_management '1'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

## Steps to Reproduce the Problem

  1. System --> OpenMPTCPRouter --> settings wizard --> Show advanced settings --> Enable IPv6 --> Enabled  
  2. Dial Modem with QMI ( IPV6 Only ).
  
 ## Specifications

  - OpenMPTCProuter version: "v0.60rc2-6.1"
  - OpenMPTCProuter VPS version: "0.1030"
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform: RPI-CM4
  - Country: INDIA
  - 
[image]

Ysurac, 2024-05-03:
Can you try to ping VPS IPv6 from both interfaces ?
If it's ok, what is the result of ```curl -k https://[<vpsip>]:65500/``` ?

iamdivyanshverma, 2024-05-03:
Yeah it's coming 
root@OpenMPTCProuter:~# curl -k https://[2400:6180:100:d0::384:d001]:65500/
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~#

Ysurac, 2024-05-03:
I would need Status->System log

iamdivyanshverma, 2024-05-03:
system log --> 
root@OpenMPTCProuter:~# logread
Jan 16 15:20:01 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Jan 16 15:20:01 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating engines.cnf
Jan 16 15:20:01 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S13openssl: Generating providers.cnf
Jan 16 15:20:02 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Jan 16 15:20:02 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Jan 16 15:20:02 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jan 16 15:20:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jan 16 15:20:02 OpenMPTCProuter authpriv.info dropbear[2357]: Not backgrounding
Jan 16 15:20:02 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: bonding
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: 8021q
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: macvlan
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: veth
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: bridge
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: Network device
Jan 16 15:20:02 OpenMPTCProuter user.notice : Added device handler type: tunnel
Jan 16 15:20:02 OpenMPTCProuter daemon.notice wpa_supplicant[2624]: Successfully initialized wpa_supplicant
Jan 16 15:20:03 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Jan 16 15:20:03 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='br-lan'
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'omr6in4' is setting up now
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Network device 'lo' link is up
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' has link connectivity 
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: wan2 (3017): The specified control device does not exist
Jan 16 15:20:03 OpenMPTCProuter daemon.notice netifd: wan1 (3016): The specified control device does not exist
[... cut ...]
May  3 10:52:36 OpenMPTCProuter daemon.warn openvpn(omr)[32645]: Use --help for more information.
May  3 10:52:41 OpenMPTCProuter daemon.warn openvpn(omr)[541]: DEPRECATED OPTION: --cipher set to 'chacha20-poly1305' but missing in --data-ciphers (AES-256-GCM:AES-128-GCM:CHACHA20-POLY1305). OpenVPN ignores --cipher for cipher negotiations. 
May  3 10:52:41 OpenMPTCProuter daemon.err openvpn(omr)[541]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)

iamdivyanshverma, 2024-05-03:
No answer to VPS IPV6 via router individual interface 
root@OpenMPTCProuter:~# ping6 -I wwan0 2400:6180:100:d0::384:d001
PING 2400:6180:100:d0::384:d001 (2400:6180:100:d0::384:d001): 56 data bytes

Ysurac, 2024-05-03:
It should.
Check that IPv6 is working on the VPS and that the correct interface is in ```/etc/shorewall6/params.net``` on the VPS

iamdivyanshverma, 2024-05-03:
root@omr-divyansh:~# ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 159.89.164.166  netmask 255.255.240.0  broadcast 159.89.175.255
        inet6 fe80::bc12:8cff:fe13:afc2  prefixlen 64  scopeid 0x20<link>
        inet6 2400:6180:100:d0::384:d001  prefixlen 64  scopeid 0x0<global>
        ether be:12:8c:13:af:c2  txqueuelen 1000  (Ethernet)
        RX packets 10004  bytes 2009490 (1.9 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6908  bytes 1727778 (1.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@omr-divyansh:~# cat /etc/shorewall6/params.net
NET_IFACE=eth0

Interface is correct

iamdivyanshverma, 2024-05-03:
Let me try with another IPV6-VPS Provider , i will update .
thanks

Ysurac, 2024-05-03:
Digital Ocean should work, I have a VPS from them.

iamdivyanshverma, 2024-05-07:
Hi @Ysurac , the issue still persists . Not able to Ping VPS IPV6 Address from individual interface.
I even tried with Modem Manager but same issue , is it possible for you to replicate at your end with a raspberry pi and a Quectel EC20/EC25 modem . Or any other modem you have .
thanks

iamdivyanshverma, 2024-05-07:
I hope this is the only script needs to run for server configuration, then followed by enabling IPV6 from VPS console.
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.1" sh

Ysurac, 2024-05-07:
You need to enable IPv6 before installing VPS script.

iamdivyanshverma, 2024-05-07:
yes i did this just now , rebuild my server enabled IPV6 first , then install using the above script

iamdivyanshverma, 2024-05-07:
one more thing i noticed , token only comes on IPV4 not with IPV6 address is it correct ?

iamdivyanshverma, 2024-05-07:
divyansh@divyansh-zorin:~$ curl --max-time 10 -s -k -H "accept: application/json" -H "Content-Type: application/x-www-form-urlencoded" -X POST -d 'username='openmptcprouter'&password='XXXXXXXXXXXXXXXXX https://2401:c080:3000:3449:5400:04ff:fee4:a614:65500/token

divyansh@divyansh-zorin:~$ curl --max-time 10 -s -k -H "accept: application/json" -H "Content-Type: application/x-www-form-urlencoded" -X POST -d 'username='openmptcprouter'&password='XXXXXXXXXXXX https://139.84.136.242:65500/token
{"access_token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiJvcGVubXB0Y3Byb3V0ZXIiLCJleHAiOjE3MTUxNjczODV9.7cx6ec6mV3lo3YZJJOvKKoHBsN6Rv94pEEymxuABJ9k","token_type":"bearer"}

Ysurac, 2024-05-07:
it's working on both but your curl command is not correct. You need to put IPv6 between [ ]

iamdivyanshverma, 2024-05-07:
curl --max-time 10 -s -k -H "accept: application/json" -H "Content-Type: application/x-www-form-urlencoded" -X POST -d 'username='openmptcprouter'&password='60C64132825A0F0A659580412EC77AF4577495FC98AA385BACB6DC6E0AA6DD3A https://[2401:c080:3000:3449:5400:04ff:fee4:a614]:65500/token

no output after putting [ ].

iamdivyanshverma, 2024-05-07:
yes

root@OMR-divyansh:~# curl -k https://[2401:c080:3000:3449:5400:04ff:fee4:a614]:65500/
"Welcome to OpenMPTCProuter Server part"

Ysurac, 2024-05-07:
Check on the VPS with ```journalctl -u omr-admin-ipv6``` but when you have the Welcome message all should be ok.

iamdivyanshverma, 2024-05-07:
root@OMR-divyansh:~# journalctl -u omr-admin-ipv6
May 07 11:20:38 OMR-divyansh systemd[1]: Started omr-admin-ipv6.service - OMR-Admin IPv6.

iamdivyanshverma, 2024-05-08:
Hi @Ysurac , I am able ping  VPS ipv6 

1.)

root@OpenMPTCProuter:~# ping 2401:c080:3000:3449:5400:4ff:fee4:a614
PING 2401:c080:3000:3449:5400:4ff:fee4:a614(2401:c080:3000:3449:5400:4ff:fee4:a614) 56 data bytes
64 bytes from 2401:c080:3000:3449:5400:4ff:fee4:a614: icmp_seq=1 ttl=52 time=48.7 ms
64 bytes from 2401:c080:3000:3449:5400:4ff:fee4:a614: icmp_seq=2 ttl=52 time=45.9 ms
64 bytes from 2401:c080:3000:3449:5400:4ff:fee4:a614: icmp_seq=3 ttl=52 time=55.1 ms
64 bytes from 2401:c080:3000:3449:5400:4ff:fee4:a614: icmp_seq=4 ttl=52 time=43.9 ms
^C
--- 2401:c080:3000:3449:5400:4ff:fee4:a614 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 43.885/48.403/55.060/4.211 ms

2.)

also getting token after running curl command manually
root@OpenMPTCProuter:~# curl --max-time 10 -s -k -H "accept: application/json" -H "Content-Typ
e: application/x-www-form-urlencoded" -X POST -d 'username='openmptcprouter'&password='XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX https://[2401:c080:3000:3449:5400:4f
f:fee4:a614]:65500/token
{"access_token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiJvcGVubXB0Y3Byb3V0ZXIiLCJleHAiOjE3MTUyMzg0Mzl9.HlBXAuT3Hv-7WjekJt8_l62wGJvVttU9gHh3XLqnwic","token_type":"bearer"}

both of the above condition happens when openmptcprouter.vps.ip='2401:c080:3000:3449:5400:04ff:fee4:a614' 
there is only IPV6 address set , adding a IPV4 here  along with IPV6 creates ping failure.

Ysurac, 2024-05-09:
You should update to latest snapshot (still compiling), there is some fix on IPv6.

iamdivyanshverma, 2024-05-09:
Thanks , i will start again with snapstop

iamdivyanshverma, 2024-05-09:
I should use 6.1 or 6.6 ?

Ysurac, 2024-05-09:
For IPv6, 6.6 is better.

iamdivyanshverma, 2024-05-09:
Alright i will move to 6.6 on both side thanks .
one more doubt suppose we have mixed WAN interfaces combination IPV4 & IPV6 will bonding work there ?

Ysurac, 2024-05-09:
With 6.6 yes, and it's really cool :)
I still have some tests/bugs to fix, this will be made next week (I need to receive some SIM cards to test all possibilities)

iamdivyanshverma, 2024-05-09:
Thanks , will test if something comes up i will update .

Ysurac, 2024-05-10:
Aggregation of IPv6 and IPv4 only work with Shadowsocks-Rust on 6.6 kernel
6.6 kernel should work, but provider seems to have blocked some VPS range, I will fork it.

iamdivyanshverma, 2024-05-10:
Okay Let me try with Shadowsocks-Rust.

one more thing , VPS Key exchange i had to do with IPV4 WAN interface once keys are exchanged then i switched all interfaces to IPV6 WAN to test Bonding which works .
still Key exchange didn't happen with IPV6 WAN . I checked with curl command got the output ,
but logread says 

May 10 12:06:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 2401:c080:3000:2c46:5400:04ff:fee7:6b5a, no server API answer on 2401:c080:3000:2c46:5400:04ff:fee7:6b5a)

closed 2024-08-14 by github-actions[bot]