Ysurac/openmptcprouter #3332: Missing Port 65280 in Documentation

Fact0ryy, 2024-05-05:
## Expected Behavior
That openmptcprouter-v0.60-6.1 works out of the box, and I can trust the documentation and have a secure system.

## Current Behavior
I am of the opinion that at least port 65280 TCP is missing in the documentation. This leads to the following two messages appearing in the router without this release on the VPS server: "ShadowSocks Rust is not running" or "Can't get public IP address from ShadowSocks Rust". As soon as I release the port, the messages disappear!

## Possible Solution
Open Port TCP/UDP 65280 in VPS

## Steps to Reproduce the Problem

  1. Setup VPS and open ports with the steps descripted in the doku (without open 65280 TCP/udp )
  2. See the result in the system status
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:
  - Kernel Version：6.1.71
  - Model：Microsoft Corporation Virtual Machine
  - Architecture：Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz
  - Firmware Version：openmptcprouter v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: Version：6.1.0-20-cloud-amd64 0.1030
  - OpenMPTCProuter VPS provider: IONOS
  - OpenMPTCProuter platform: VM in Hyper-V
  - Country: europe

Ysurac, 2024-05-05:
Added, and XRay/V2Ray ports also added in the doc.
It's easier to open range 65000-65535
Thanks

closed 2024-05-05 by Ysurac

Fact0ryy, 2024-05-05:
Thank you for the update in the documentation. I noticed a small typo in the listing:

Xray port is [...], 6525**5**1, [..] VLESS-Reality is used (TCP)

Fact0ryy, 2024-05-05:
> Added, and XRay/V2Ray ports also added in the doc. It's easier to open range 65000-65535 Thanks

Thank you for the update in the documentation. I noticed a small typo in the listing:

Xray port is [...], 652551, [..] VLESS-Reality is used (TCP)

I suspect that a 5 is too much.

Ysurac, 2024-05-05:
fixed, thanks