Ysurac/openmptcprouter #3334: Slow speed of mptcprouter

MistiqueX, 2024-05-06:
i use raspberry 4pi + 3 em12-g and openmptcprouter v0.60-6.1 r0+24843-acf40c022e
1 - i cant have good speed, its not summ all modems and speed is to slow. - the problem on VPS server? or configuration of MPTCP?
2 - no way to aggregate frequency's and 3 modem working at once - on windows computer 1 modem agregate 100-200 mbit/s, on mptcprouet - 100 on 3 modems.....

OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
OpenMPTCProuter VPS version: 060.-6.1
OpenMPTCProuter VPS provider: timeweb
OpenMPTCProuter platform: RPI4/

Ysurac, 2024-05-06:
Do you have a powered USB hub ?
Modems are on several providers ?
Can you test with ```omr-test-speed <interface>``` via SSH each interfaces and even 2 at same time ?
Can you test with  ```omr-test-speed``` to test aggregated speed (Ctrl+C to stop after 2 minutes) ?
All is green in status page ?

MistiqueX, 2024-05-07:
Do you have a powered USB hub ?    - YES
Modems are on several providers ?   - YES  - 2 same, 1 another.
All is green in status page ?    -YES
[image]

Can you test with omr-test-speed <interface> via SSH each interfaces and even 2 at same time ?
Can you test with omr-test-speed to test aggregated speed (Ctrl+C to stop after 2 minutes) ? 
will do tomorow.

ccmks, 2024-05-09:
Per my experience, having multiple GSM modem that each of them have same ISP would not be benefit for aggregation. You need to have multiple GSM provider in order to get the MPTCP aggregation, or if you are still insisting of doing same GSM provider, you need to be able to have each modem to get signal from separate tower (usually done by pointing 2 antennas to 2 different tower)

MistiqueX, 2024-05-11:
> Per my experience, having multiple GSM modem that each of them have same ISP would not benefit for aggregation. You need to have multiple GSM provider in order to get the MPTCP aggregation, or if you are still insisting of doing same GSM provider, you need to be able to have each modem to get signal from separate tower (usually done by pointing 2 antennas to 2 different tower)

You're wrong. The operator makes it possible to aggregate each SIM card, no matter which device it is located in, the extra device would aggregate. Therefore, I ask you not to mislead people! This will have a very detrimental effect on you in the future.
I'll tell you a secret - summation without a VPS server is also possible, it is also implemented as bounding by microtik and other methods, and I'm talking about the 1st operator and the 1st tower. In my practice, I already have 2 b818 huawey devices and I'm going to add 1 more.

Ysurac, 2024-05-12:
@MistiqueX There theory and practice.
In many case having multiple modems with all SIMs cards on same operator doesn't help (due to many factor, like cell speed limit, the fact that all modems are in same place,... The problem can be different if there is more than one cell available not using limited microwave relay link)
For the bounding without VPS, it's only load balancing. OpenMPTCProuter do the same when there is no VPS. So no real aggregation and different output IPs.
Aggregation is also possible when website support MPTCP, but there is not many for now.

MistiqueX, 2024-05-13:
Hi there.
BusyBox v1.33.2 (2022-05-17 22:11:28 UTC) built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r0+16594-ce92de8c8c)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.59.1
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Mon Aug 29 20:38:49 UTC 2022
-----------------------------------------------------
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~# host: lax.download.datapacket.com - ping: 221
-ash: host:: not found
root@OpenMPTCProuter:~# omr-test-speed wwan2
Select best test server...
host: scaleway.testdebit.info - ping:
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 63
host: lyon.testdebit.info - ping: 61
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 72
host: appliwave.testdebit.info - ping: 60
host: speedtest.frankfurt.linode.com - ping: 51
host: speedtest.tokyo2.linode.com - ping: 332
host: speedtest.singapore.linode.com - ping: 371
host: speedtest.newark.linode.com - ping: 132
host: speedtest.atlanta.linode.com - ping: 157
host: speedtest.dallas.linode.com - ping: 176
host: speedtest.fremont.linode.com - ping: 207
host: speed.hetzner.de - ping: 53
host: ipv4.bouygues.testdebit.info - ping: 61
host: par.download.datapacket.com - ping: 65
host: nyc.download.datapacket.com - ping: 149
host: ams.download.datapacket.com - ping: 63
host: fra.download.datapacket.com - ping: 57
host: lon.download.datapacket.com - ping: 59
host: mad.download.datapacket.com - ping:
host: prg.download.datapacket.com - ping: 78
host: sto.download.datapacket.com - ping: 70
host: vie.download.datapacket.com - ping: 73
host: war.download.datapacket.com - ping: 41
host: atl.download.datapacket.com - ping: 160
host: chi.download.datapacket.com - ping: 159
host: lax.download.datapacket.com - ping: 216
host: mia.download.datapacket.com - ping: 192
host: nyc.download.datapacket.com - ping: 141
host: speedtest.milkywan.fr - ping: 60
Best server is http://war.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:01:15 --:--:--     0^C
root@OpenMPTCProuter:~# omr-test-speed wwan3
You must use a real interface. You wan find them using 'ip a' for example
root@OpenMPTCProuter:~# omr-test-speed wwan3
You must use a real interface. You wan find them using 'ip a' for example
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~# omr-test-speed wwan3
[... 98 lines cut ...]
host: fra.download.datapacket.com - ping: 59
host: lon.download.datapacket.com - ping: 71
host: mad.download.datapacket.com - ping:
host: prg.download.datapacket.com - ping: 65
host: sto.download.datapacket.com - ping: 79
host: vie.download.datapacket.com - ping: 60
host: war.download.datapacket.com - ping: 69
host: atl.download.datapacket.com - ping: 157
host: chi.download.datapacket.com - ping: 167
host: lax.download.datapacket.com - ping: 216
host: mia.download.datapacket.com - ping: 183
host: nyc.download.datapacket.com - ping: 200
host: speedtest.milkywan.fr - ping: 71
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  188M    0     0  10.8M      0  0:14:40  0:00:17  0:14:23 12.0M
curl: (18) transfer closed with 9802353481 bytes remaining to read

root@OpenMPTCProuter:~#

This is when modem is connected to pc
[image]

i think the mptcprouter - cant agregate normal speed.

Ysurac, 2024-05-13:
There is only omr-test-speed wwan0 in your result, can you test with wwan1 ? and wwan2 again (it's failing here).
It's also possible the website used for speed test in omr-test-speed is not very good...
In the short aggregated test, without the interface, it seems to use another server and give 12.0Mo/s so 96Mbps
You may try also to test aggregated speed with 2 modems first.

And try to update the router part, you are use 0.59.1 in the screenshot.

MistiqueX, 2024-05-13:
> There is only omr-test-speed wwan0 in your result, can you test with wwan1 ? and wwan2 again (it's failing here). It's also possible the website used for speed test in omr-test-speed is not very [...]

root@OpenMPTCProuter:~# omr-test-speed wwan1
Select best test server...
host: scaleway.testdebit.info - ping:
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 89
host: lyon.testdebit.info - ping: 58
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 75
host: appliwave.testdebit.info - ping: 58
host: speedtest.frankfurt.linode.com - ping: 64
host: speedtest.tokyo2.linode.com - ping: 312
host: speedtest.singapore.linode.com - ping: 382
host: speedtest.newark.linode.com - ping: 142
host: speedtest.atlanta.linode.com - ping: 165
host: speedtest.dallas.linode.com - ping: 187
host: speedtest.fremont.linode.com - ping: 205
host: speed.hetzner.de - ping: 63
host: ipv4.bouygues.testdebit.info - ping: 63
host: par.download.datapacket.com - ping: 66
host: nyc.download.datapacket.com - ping: 154
host: ams.download.datapacket.com - ping: 62
host: fra.download.datapacket.com - ping: 68
host: lon.download.datapacket.com - ping: 72
host: mad.download.datapacket.com - ping:
host: prg.download.datapacket.com - ping: 69
host: sto.download.datapacket.com - ping: 88
host: vie.download.datapacket.com - ping: 81
host: war.download.datapacket.com - ping: 38
host: atl.download.datapacket.com - ping: 181
host: chi.download.datapacket.com - ping: 150
host: lax.download.datapacket.com - ping: 227
host: mia.download.datapacket.com - ping: 183
host: nyc.download.datapacket.com - ping: 150
host: speedtest.milkywan.fr - ping: 75
Best server is http://war.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:15 --:--:--     0
curl: (52) Empty reply from server

root@OpenMPTCProuter:~# omr-test-speed wwan2
Select best test server...
host: scaleway.testdebit.info - ping:
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 67
host: lyon.testdebit.info - ping: 60
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping: 70
host: appliwave.testdebit.info - ping: 59
host: speedtest.frankfurt.linode.com - ping: 49
host: speedtest.tokyo2.linode.com - ping: 314
host: speedtest.singapore.linode.com - ping: 372
host: speedtest.newark.linode.com - ping: 129
host: speedtest.atlanta.linode.com - ping: 176
host: speedtest.dallas.linode.com - ping: 174
host: speedtest.fremont.linode.com - ping: 212
host: speed.hetzner.de - ping: 53
host: ipv4.bouygues.testdebit.info - ping: 61
host: par.download.datapacket.com - ping: 68
host: nyc.download.datapacket.com - ping: 144
host: ams.download.datapacket.com - ping: 54
host: fra.download.datapacket.com - ping: 56
host: lon.download.datapacket.com - ping: 70
host: mad.download.datapacket.com - ping:
host: prg.download.datapacket.com - ping: 61
host: sto.download.datapacket.com - ping: 64
host: vie.download.datapacket.com - ping: 64
host: war.download.datapacket.com - ping: 38
[... 143 lines cut ...]
host: speedtest.atlanta.linode.com - ping: 154
host: speedtest.dallas.linode.com - ping: 157
host: speedtest.fremont.linode.com - ping: 231
host: speed.hetzner.de - ping: 56
host: ipv4.bouygues.testdebit.info - ping: 57
host: par.download.datapacket.com - ping: 55
host: nyc.download.datapacket.com - ping: 136
host: ams.download.datapacket.com - ping: 58
host: fra.download.datapacket.com - ping: 205
host: lon.download.datapacket.com - ping: 64
host: mad.download.datapacket.com - ping:
host: prg.download.datapacket.com - ping: 86
host: sto.download.datapacket.com - ping: 43
host: vie.download.datapacket.com - ping: 73
host: war.download.datapacket.com - ping: 114
host: atl.download.datapacket.com - ping: 152
host: chi.download.datapacket.com - ping: 169
host: lax.download.datapacket.com - ping: 212
host: mia.download.datapacket.com - ping: 182
host: nyc.download.datapacket.com - ping: 144
host: speedtest.milkywan.fr - ping: 82
Best server is http://sto.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 12 9536M   12 1202M    0     0  4473k      0  0:36:23  0:04:35  0:31:48 5918k

MistiqueX, 2024-05-13:
[image]

Ysurac, 2024-05-13:
Seems that it's not able to start glorytun-tcp. Try a "systemctl stop glorytun-tcp@tun0 && apt-get remove omr-glorytun-tcp"
You are using Ubuntu ?

MistiqueX, 2024-05-13:
> Seems that it's not able to start glorytun-tcp. Try a "systemctl stop glorytun-tcp@tun0 && apt-get remove omr-glorytun-tcp" You are using Ubuntu ?

Yes UBUNTU

root@2236871-cl74986:~# systemctl stop glorytun-tcp@tun0 && apt-get remove omr-glorytun-tcp
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  apport-symptoms eatmydata libeatmydata1 python3-blinker python3-entrypoints python3-httplib2 python3-importlib-metadata
  python3-jinja2 python3-json-pointer python3-jsonpatch python3-jsonschema python3-keyring python3-lazr.uri python3-markupsafe
  python3-more-itertools python3-problem-report python3-pyrsistent python3-secretstorage python3-simplejson python3-systemd
  python3-wadllib python3-zipp
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  omr-glorytun-tcp
0 upgraded, 0 newly installed, 1 to remove and 8 not upgraded.
1 not fully installed or removed.
After this operation, 59.4 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 106995 files and directories currently installed.)
Removing omr-glorytun-tcp (0.0.35-6) ...

MistiqueX, 2024-05-14:
Whats next BOSS?

MistiqueX, 2024-05-22:
> Seems that it's not able to start glorytun-tcp. Try a "systemctl stop glorytun-tcp@tun0 && apt-get remove omr-glorytun-tcp" You are using Ubuntu ?

[image]
Can u help me ?

MistiqueX, 2024-05-22:
no way to good speed (((

Ysurac, 2024-05-22:
Use Debian on VPS, it's the easiest way for now.

MistiqueX, 2024-05-22:
root@2236871-cl74986:~# omr-test-speed
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 48
host: speedtest.tokyo2.linode.com - ping: 317
host: speedtest.singapore.linode.com - ping: 388
host: speedtest.newark.linode.com - ping: 124
host: speedtest.atlanta.linode.com - ping: 139
host: speedtest.dallas.linode.com - ping: 159
host: speedtest.fremont.linode.com - ping: 190
host: speedtest.tele2.net - ping: 49
host: speed.hetzner.de - ping: 39
host: ipv4.bouygues.testdebit.info - ping: 44
host: par.download.datapacket.com - ping: 51
host: nyc.download.datapacket.com - ping: 120
host: ams.download.datapacket.com - ping: 46
host: fra.download.datapacket.com - ping: 39
host: lon.download.datapacket.com - ping: 54
host: mad.download.datapacket.com - ping: 
host: prg.download.datapacket.com - ping: 47
host: sto.download.datapacket.com - ping: 77
host: vie.download.datapacket.com - ping: 52
host: war.download.datapacket.com - ping: 65
host: atl.download.datapacket.com - ping: 140
host: chi.download.datapacket.com - ping: 139
host: lax.download.datapacket.com - ping: 191
host: mia.download.datapacket.com - ping: 154
host: nyc.download.datapacket.com - ping: 120
Best server is https://speed.hetzner.de/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   190  100   190    0     0    984      0 --:--:-- --:--:-- --:--:--   979

Ysurac, 2024-05-22:
This can't work in current configuration as kernel is not the same on Router and VPS.

MistiqueX, 2024-05-22:
> ot the same on Router and
can u help with that ? - which command is need to upgrade VPS?

MistiqueX, 2024-05-22:
patch is already the newest version (2.7.6-6).
pkg-config is already the newest version (0.29.1-0ubuntu4).
rename is already the newest version (1.10-1).
curl is already the newest version (7.68.0-1ubuntu2.22).
dirmngr is already the newest version (2.2.19-3ubuntu2.2).
ipset is already the newest version (7.5-1ubuntu0.20.04.1).
libcurl4 is already the newest version (7.68.0-1ubuntu2.22).
unzip is already the newest version (6.0-25ubuntu1.2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

MistiqueX, 2024-05-22:
[image]
Dear Author! how i can have good speed?

root@2236871-cl74986:~# omr-test-speed
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 45
host: speedtest.tokyo2.linode.com - ping: 326
host: speedtest.singapore.linode.com - ping: 396
host: speedtest.newark.linode.com - ping: 125
host: speedtest.atlanta.linode.com - ping: 139
host: speedtest.dallas.linode.com - ping: 159
host: speedtest.fremont.linode.com - ping: 190
host: speedtest.tele2.net - ping: 49
host: speed.hetzner.de - ping: 39
host: ipv4.bouygues.testdebit.info - ping: 44
host: par.download.datapacket.com - ping: 51
host: nyc.download.datapacket.com - ping: 119
host: ams.download.datapacket.com - ping: 48
host: fra.download.datapacket.com - ping: 39
host: lon.download.datapacket.com - ping: 54
host: mad.download.datapacket.com - ping: 
host: prg.download.datapacket.com - ping: 49
host: sto.download.datapacket.com - ping: 77
host: vie.download.datapacket.com - ping: 53
host: war.download.datapacket.com - ping: 65
host: atl.download.datapacket.com - ping: 142
host: chi.download.datapacket.com - ping: 139
host: lax.download.datapacket.com - ping: 196
host: mia.download.datapacket.com - ping: 154
host: nyc.download.datapacket.com - ping: 119
Best server is https://speed.hetzner.de/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   190  100   190    0     0    920      0 --:--:-- --:--:-- --:--:--   926

Ysurac, 2024-05-22:
Good kernel this time.
Can you test with nperf for example ?
Hetzner speed file is down, so the test don't work. It's fixed in snapshot so will be in next release.

MistiqueX, 2024-05-22:
> It's 23.2Mo/s, so 23.2*8=185.6Mb/s On the screenshot I see that the RPI4 have a quite high load average. A RPI4 can get up to 550Mb/s. You should check Status->System log on the router.
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_server)[9124]: library versions: OpenSSL 3.0.12 24 Oct 2023, LZO 2.10
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_server)[9124]: DCO version: N/A
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_client)[9125]: OpenVPN 2.6.10 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_client)[9125]: library versions: OpenSSL 3.0.12 24 Oct 2023, LZO 2.10
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_server)[9124]: net_route_v4_best_gw query: dst 0.0.0.0
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_client)[9125]: DCO version: N/A
May 22 18:32:20 OpenMPTCProuter daemon.warn openvpn(sample_client)[9125]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
May 22 18:32:20 OpenMPTCProuter daemon.warn openvpn(sample_client)[9125]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
May 22 18:32:20 OpenMPTCProuter daemon.warn openvpn(sample_client)[9125]: OpenSSL: error:80000002:system library::No such file or directory:calling fopen(/etc/openvpn/client.crt, r)
May 22 18:32:20 OpenMPTCProuter daemon.warn openvpn(sample_client)[9125]: OpenSSL: error:10000080:BIO routines::no such file:
May 22 18:32:20 OpenMPTCProuter daemon.warn openvpn(sample_client)[9125]: OpenSSL: error:0A080002:SSL routines::system lib:
May 22 18:32:20 OpenMPTCProuter daemon.err openvpn(sample_client)[9125]: Cannot load certificate file /etc/openvpn/client.crt
May 22 18:32:20 OpenMPTCProuter daemon.notice openvpn(sample_client)[9125]: Exiting due to fatal error
May 22 18:32:23 OpenMPTCProuter daemon.warn openvpn(sample_server)[9124]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
May 22 18:32:23 OpenMPTCProuter daemon.warn openvpn(sample_server)[9124]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
May 22 18:32:23 OpenMPTCProuter daemon.err openvpn(sample_server)[9124]: OpenSSL: error:80000002:system library::No such file or directory:calling fopen(/etc/openvpn/dh2048.pem, r)
May 22 18:32:23 OpenMPTCProuter daemon.err openvpn(sample_server)[9124]: OpenSSL: error:10000080:BIO routines::no such file:
May 22 18:32:23 OpenMPTCProuter daemon.err openvpn(sample_server)[9124]: Cannot open /etc/openvpn/dh2048.pem for DH parameters
May 22 18:32:23 OpenMPTCProuter daemon.notice openvpn(sample_server)[9124]: Exiting due to fatal error
May 22 18:32:25 OpenMPTCProuter daemon.err openvpn(custom_config)[9586]: Options error: In [CMD-LINE]:1: Error opening configuration file: my-vpn.conf
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(custom_config)[9586]: Use --help for more information.
May 22 18:32:25 OpenMPTCProuter daemon.notice openvpn(sample_client)[9592]: Note: --cipher is not set. OpenVPN versions before 2.5 defaulted to BF-CBC as fallback when cipher negotiation failed in this case. If you need this fallback please add '--data-ciphers-fallback BF-CBC' to 
May 22 18:32:25 OpenMPTCProuter daemon.notice openvpn(sample_client)[9592]: OpenVPN 2.6.10 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
May 22 18:32:25 OpenMPTCProuter daemon.notice openvpn(sample_client)[9592]: library versions: OpenSSL 3.0.12 24 Oct 2023, LZO 2.10
May 22 18:32:25 OpenMPTCProuter daemon.notice openvpn(sample_client)[9592]: DCO version: N/A
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(sample_client)[9592]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(sample_client)[9592]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(sample_client)[9592]: OpenSSL: error:80000002:system library::No such file or directory:calling fopen(/etc/openvpn/client.crt, r)
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(sample_client)[9592]: OpenSSL: error:10000080:BIO routines::no such file:
May 22 18:32:25 OpenMPTCProuter daemon.warn openvpn(sample_client)[9592]: OpenSSL: error:0A080002:SSL routines::system lib:
May 22 18:32:25 OpenMPTCProuter daemon.err openvpn(sample_client)[9592]: Cannot load certificate file /etc/openvpn/client.crt
May 22 18:32:25 OpenMPTCProuter daemon.notice openvpn(sample_client)[9592]: Exiting due to fatal error
May 22 18:32:28 OpenMPTCProuter daemon.warn openvpn(sample_server)[10136]: WARNING: --topology net30 support for server configs with IPv4 pools will be removed in a future release. Please migrate to --topology subnet as soon as possible.
May 22 18:32:28 OpenMPTCProuter daemon.notice openvpn(sample_server)[10136]: Note: --cipher is not set. OpenVPN versions before 2.5 defaulted to BF-CBC as fallback when cipher negotiation failed in this case. If you need this fallback please add '--data-ciphers-fallback BF-CBC' to
May 22 18:32:28 OpenMPTCProuter daemon.notice openvpn(sample_server)[10136]: Note: NOT using '--topology subnet' disables data channel offload.
[... cut ...]
May 22 18:33:15 OpenMPTCProuter daemon.notice openvpn(sample_client)[29522]: Exiting due to fatal error
May 22 18:33:15 OpenMPTCProuter daemon.err openvpn(sample_server)[29523]: OpenSSL: error:10000080:BIO routines::no such file:
May 22 18:33:15 OpenMPTCProuter daemon.err openvpn(sample_server)[29523]: Cannot open /etc/openvpn/dh2048.pem for DH parameters
May 22 18:33:15 OpenMPTCProuter daemon.notice openvpn(sample_server)[29523]: Exiting due to fatal error
May 22 18:33:18 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifupdate of wan2 (wwan0)

Ysurac, 2024-05-22:
There is a problem with OpenVPN: example seems to be enabled. Did you change OpenVPN config ?

MistiqueX, 2024-05-22:
> There is a problem with OpenVPN: example seems to be enabled. Did you change OpenVPN config ?

[image]
i try to click some, but it was just once

Ysurac, 2024-05-23:
You seems to have no VPN enabled in wizard and omrvpn interface seems to be a bridge with DHCP client enabled. Should be unmanaged on device tun0.

MistiqueX, 2024-05-23:
> You seems to have no VPN enabled in wizard

omrvpn interface with tun0 is working. VPN - is glotrum tcp. 
mow is 1 problem - slow speed!

MistiqueX, 2024-05-23:
Dear Ysurac - no way to have a good speed... very sad, to much time and so bad result....maybe its my fault and i dont know what to do......20 mbps - so slow

Ysurac, 2024-05-23:
Did you test ```omr-test-speed-server vps wwan0``` on each interfaces ? To test speed between connection and server ?
Did you try to disconnect interfaces one by one ?
Do you now get a good status page with low load average and no more dirty logs ?
What is the latency between each connection and the server ? ```ping -B -I wwan0 <serverip>``` to check on wwan0
And remember that omr-test-speed give the speed in MB/s not Mbps.

ccmks, 2024-05-27:
> You're wrong. The operator makes it possible to aggregate each SIM card, no matter which device it is located in, the extra device would aggregate. Therefore, I ask you not to mislead people! This [...]

Like I said, per my experience that's what happened regardless you are telling me right or wrong, it wouldn't change the fact that this practice is happening.

We are living in developed country where bandwidth is very scarce. Imagine that each tower on outside city area only equipment with 100 Mbps bandwidth link, and everyone is trying to get connected to it. By adding more device means you are adding the crowd to the crowded uplink on the tower. How is it going to make it faster?! On the top of that, if the band channel is crowded too, it would make it worse.

So like I said before, this is based on my experience and of course your experience won't be the same so in this case I am not providing any misleading information! I was giving the insight that having multiple devices connected to same tower provider on same band usually doesn't get much benefit from doing aggregation.

If you are getting benefit, that means the uplink bandwidth on the tower is pretty big and probably not at full capacity. Be blessed if you have that condition!

closed 2024-08-30 by github-actions[bot]