Ysurac/openmptcprouter #3335: Proxmox and OMR poor Performance ...

vinceducat, 2024-05-06:
## Expected Behavior
Have OMR router on a proxmox pve 8.2 with performance as in direct connection to the ISP Box

## Current Behavior
throught proxmox and OMR as guest, i have half of the nominal speed on ma connexion "Fiber"

direct link from PromoxPVE  to freebox 1.6Gb/s => 200Mo/s
from OMR itself  80-100Mo/s :-(

## Specifications

Promox PVE machine :  ms-01 miniforum with i9 -13 proc 96Go ram and 3 ssd m.2 in zfs 
vmbr0 is link to 10Gb/s port
vmbr1 is link  to the Freebox at 2.5Gb/b port

Freebox Pop with  2.5Gb port connect to proxmox with also 2.5Gb/s port
Guest OMR VM on proxmox have 2 NIC virtio 1 for lan

needs help to know il it is a problem of OMR or proxmox ...

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 0.1310
  - OpenMPTCProuter VPS provider: milkywan
  - OpenMPTCProuter platform: x86_64 as guest on promox -->

Ysurac, 2024-05-07:
Did you can try e1000 instead of virtio ?

Scaff31, 2024-05-08:
[image]

Which offer do you use ?

vinceducat, 2024-05-08:
@Ysurac yes a try E1000 and virtio.

@Scaff31 i have my own vps on milkywan not link to mega or giga offer.

@Ysurac  et scaff31 ... after many test the problème that in fact Freebox Pop "filter" ipv4 traffic versus ip v6.

i done direct connection beetween a computer with a 2.5Gb/s and the freebox pop link on the 2.5Gb/s port...

the result in ipv4 is 700/300 and 2000/700 in ipv 6.... :-( try with different machine et serveur of test.

@Ysurac is it possible in omr router to use only ipv6 fromm OMR router to the OMR Vps ? => i would try to bypass the problem

Ysurac, 2024-05-08:
Yes it's possible, less tested for now but should work.

vinceducat, 2024-05-08:
@Ysurac  do you have some sample for setup omr router .. in the wizard ?

vps is milkywan i have ipv4 adresse
isp is Free with freebox pop.. (FTTH) (can i let freebox in router mode or put it i bridge mode ?)

you will welcome if you have some tips/samples ;-)

do i need to check some thinks on the omr vps side

Ysurac, 2024-05-08:
Enable IPv6 via the "Advanced Settings" checkbox, and set an IPv6 address.
OMR also support IPv4 and IPv6 for same server and will use both and can even aggregate IPv6 connections with IPv4 connections with kernel 6.6 on snapshot.
Snapshot solve some issues with IPv6, there is still some that will be solved soon.

On VPS side, IPv6 need to be working before VPS install script is run.

vinceducat, 2024-05-08:
@Ysurac  at this time  : 
ipv6 is enable by the check box and the address is fd58:8a43:2f89::/48

for vps i have ipv4 address of the milkywan vps do i add the ipv6 beside or remplace it ?

vinceducat, 2024-05-08:
@Ysurac i try to make test with on new vps on hetner on debian 12 with snapshot version 0.61 6.6 but i have this error
--2024-05-08 20:56:17--  https://dl.xanmod.org/archive.key
Resolving dl.xanmod.org (dl.xanmod.org)... 2a06:98c1:3120::3, 2a06:98c1:3121::3, 188.114.96.3, ...
Connecting to dl.xanmod.org (dl.xanmod.org)|2a06:98c1:3120::3|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2024-05-08 20:56:17 ERROR 403: Forbidden.

any idea how to solve that

vinceducat, 2024-05-08:
try a second time 

install and have 

Err:10 http://mirror.hetzner.com/debian/security bookworm-security/main amd64 Packages
  404  Not Found [IP: 2a01:4ff:ff00::3:3 80]
Ign:10 http://mirror.hetzner.com/debian/security bookworm-security/main amd64 Packages
Err:10 http://mirror.hetzner.com/debian/security bookworm-security/main amd64 Packages
  404  Not Found [IP: 2a01:4ff:ff00::3:3 80]
Reading package lists... Done
E: Failed to fetch http://mirror.hetzner.com/debian/security/dists/bookworm-security/main/binary-amd64/Packages  404  Not Found [IP: 2a01:4ff:ff00::3:3 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.

vinceducat, 2024-05-08:
@Ysurac the script kill the vm after a restart ... i have to reinstall ... :-(

Ysurac, 2024-05-09:
It seems that xanmod have an issue. I will check...

vinceducat, 2024-05-09:
@Ysurac  dl.xanmod.org seems to be not accessible from hetzner but ok from my home ... :-(

vinceducat, 2024-05-13:
@Ysurac install of test version is ok now ..

closed 2024-08-16 by github-actions[bot]