Ysurac/openmptcprouter #3336: General issues in the log and status in the latest snapshots.

vempire-ghost, 2024-05-07:
## Expected Behavior
Everything works.

## Current Behavior
Some erros in log and in status window.

## Specifications

  - OpenMPTCProuter version: General issues in the log and status in the latest snapshots. snap from 06/05
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1031-test
  - OpenMPTCProuter VPS provider: OCI
  - OpenMPTCProuter platform: x86_64
  - Country: 

Here some logs.
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: BusyBox v1.33.2 (2023-07-29 19:00:07 UTC) multi-call binary.
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: Usage: grep [-HhnlLoqvsrRiwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: Search for PATTERN in FILEs (or stdin)
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-H	Add 'filename:' prefix
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-h	Do not add 'filename:' prefix
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-n	Add 'line_no:' prefix
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-l	Show only names of files that match
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-L	Show only names of files that don't match
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-c	Show only count of matching lines
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-o	Show only the matching part of line
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-q	Quiet. Return 0 if PATTERN is found, 1 otherwise
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-v	Select non-matching lines
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-s	Suppress open and read errors
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-r	Recurse
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-R	Recurse and dereference symlinks
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-i	Ignore case
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-w	Match whole words only
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-x	Match whole lines only
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-F	PATTERN is a literal (not regexp)
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-E	PATTERN is an extended regexp
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-m N	Match up to N times per file
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-A N	Print N lines of trailing context
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-B N	Print N lines of leading context
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-C N	Same as '-A N -B N'
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-e PTRN	Pattern to match
May  7 02:28:26 OpenMPTCProuter daemon.err omr-tracker-server[13857]: 	-f FILE	Read pattern from file

and

May  7 02:28:48 OpenMPTCProuter daemon.err glorytun[10488]: 140.238.178.119.65001: connected
May  7 02:28:48 OpenMPTCProuter daemon.info glorytun[10488]: STARTED tun0
May  7 02:29:00 OpenMPTCProuter user.notice MPTCP: Set tun0 to off from tun0 is in default mode
May  7 02:29:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
May  7 02:29:01 OpenMPTCProuter user.notice omr-tracker: Launched
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0 (was 192.168.0.1)
May  7 02:29:02 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
May  7 02:29:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
May  7 02:29:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan3 (eth3) switched off
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan5 (eth5) switched off
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth4) switched off
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth3
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan1 (eth1) switched off
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth5
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth2
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth4
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth1
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth3
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth5
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth2
May  7 02:29:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth4
[... 60 lines cut ...]
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan3 (eth3) switched off
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth3
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan5 (eth5) switched off
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth4) switched off
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth5
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth3
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth4
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth2
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth5
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth4
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth2
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan1 (eth1) switched off
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth1
May  7 02:29:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 140.238.178.119 dev eth1
May  7 02:29:51 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
May  7 02:29:51 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
May  7 02:29:52 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
May  7 02:29:53 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
May  7 02:30:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
May  7 02:30:02 OpenMPTCProuter user.notice firewall.omr-bypass: reload omr-bypass rules
May  7 02:30:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
May  7 02:30:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running

[image]

Ysurac, 2024-05-08:
Maybe 1st log is solved.
For the second part, I just added more log. When new snapshot will be compiled this will help to see why interfaces are switched off.

vempire-ghost, 2024-05-08:
Thanks, Ysurac. The first log was really fixed. Here's the new log for the second problem. What I find odd is that OMR is referencing the VPS IP instead of the WAN IP in the route it creates and deletes.

May  8 22:34:50 OpenMPTCProuter daemon.err glorytun[12277]: MY VPS IP.65001: connected
May  8 22:34:50 OpenMPTCProuter daemon.info glorytun[12277]: STARTED tun0
May  8 22:34:51 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0 (was 192.168.0.1)
May  8 22:34:51 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later
May  8 22:34:51 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
May  8 22:35:06 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
May  8 22:35:06 OpenMPTCProuter user.notice omr-tracker: Launching...
May  8 22:35:20 OpenMPTCProuter user.notice omr-tracker: Launched
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan3 (eth3) switched off (interface up status: true - Tracker status: OK)
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth4) switched off (interface up status: true - Tracker status: OK)
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan5 (eth5) switched off (interface up status: true - Tracker status: OK)
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth3
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth4
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth5
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan1 (eth1) switched off (interface up status: true - Tracker status: OK)
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth1
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth2) switched off (interface up status: true - Tracker status: OK)
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth2
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth4
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth1
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth3
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth5
May  8 22:35:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to MY VPS IP dev eth2
May  8 22:35:23 OpenMPTCProuter daemon.info dnsmasq[18427]: read /etc/hosts - 4 addresses
May  8 22:35:23 OpenMPTCProuter daemon.info dnsmasq[18427]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
May  8 22:35:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[18427]: read /etc/ethers - 0 addresses
May  8 22:35:23 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
May  8 22:35:23 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
May  8 22:35:26 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
May  8 22:35:26 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
May  8 22:35:26 OpenMPTCProuter user.notice omr-tracker: Launching...
May  8 22:35:30 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is down
May  8 22:35:30 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity loss
May  8 22:35:32 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is up
May  8 22:35:32 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity

Ysurac, 2024-05-09:
For the route it's normal, there is a direct route to the VPS from each connection.
For the switched off interface, I will check.

Ysurac, 2024-05-09:
The problem will be fixed in next snapshot, it's compiling.

vempire-ghost, 2024-05-10:
In the latest snapshot, the problem didn't occur anymore. Thank you very much.

closed 2024-05-10 by vempire-ghost