Ysurac/openmptcprouter #3344: Project stream SRT box

esquilo444, 2024-05-10:
<Good morning ,

I wanted to know if there was a project underway to optimize the configuration of the SRT video transmission protocol?
I know that the subject has already been discussed a few times, without any real solution found. UDP multiparh is the heart of the problem...
I found nothing that was less than two years old on the open mtpcp router forum.
I have high hopes for the next version with kernel 6.6 or 6.7 or 6.8...
I encounter the same problems as all the people who use this protocol... so of course I tried all possible combinations of proxy and VPN, with or without etc... I am on a project for stream retransmissions video for the regional France 3 in my region...
Many thanks in advance.
Jerome

## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: <! version developer 6.1- -->
  - OpenMPTCProuter VPS provider:ovh
  - OpenMPTCProuter platform:

esquilo444, 2024-05-10:
I forget thé router device.. RPI4😅

esquilo444, 2024-05-11:
Since I use open mtpcp router on tcp traffic, it's great. But for my SRT video streams, it is UDP that interests me and on this point the different possible solutions work poorly or not at all for me... The most coherent choice would be according to what I have read shadowsock as proxy for tcp and glorytun UDP to aggregate the UDP stream in the vpn.. Unfortunately glorytun udp never worked for me for sending srt streams. The rtt is correct, but the flow instantly loses more than 50 percent of the data... vps on ovh and router on RPI 4..
maybe it's my hardware... the only approximately functional solution is V2RAY by activating the udp stream by v2ray... but I am limited in bitrate, because if I increase the bitrate of the v2ray video stream it is quickly less efficient: packet loss increases, etc.
Thank you in advance for your lights.

Jerome

Ysurac, 2024-05-12:
Nothing new on UDP side for now.
I will work on it asap.

esquilo444, 2024-05-12:
I don't no why, i can't us glorytun udp . It doesn't work for me ...

esquilo444, 2024-05-12:
With shadowsok a'd v2ray ... Same problèm ...

closed 2024-08-16 by github-actions[bot]