Ysurac/openmptcprouter #3347: "OpenMPTCPRouter Status" sometimes stops displaying the latency of the WANs.

vempire-ghost, 2024-05-11:
## Expected Behavior
"OpenMPTCPRouter Status" consistently displays the latency of the WANs.

## Current Behavior
Randomly, the OpenMPTCPRouter status screen stops displaying the latencies, and then resumes displaying them without any apparent reason. Initially, it doesn't show them upon connecting to the VPS, but sometimes it starts displaying them and then stops after a while. Conversely, sometimes it doesn't show them initially, but then starts displaying them later on.

[image]
May 10 22:13:16 OpenMPTCProuter daemon.err glorytun[14925]: xxx.238.178.xxx.65001: connected
May 10 22:13:16 OpenMPTCProuter daemon.info glorytun[14925]: STARTED tun0
May 10 22:13:16 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0 (was 192.168.0.1)
May 10 22:13:20 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
May 10 22:13:22 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (xxx.238.178.xxx)
May 10 22:13:38 OpenMPTCProuter daemon.info unbound: [30093:0] info: service stopped (unbound 1.17.1).
May 10 22:13:38 OpenMPTCProuter daemon.info unbound: [30093:0] info: server stats for thread 0: 0 queries, 0 answers from cache, 0 recursions, 0 prefetch, 0 rejected by ip ratelimiting
May 10 22:13:38 OpenMPTCProuter daemon.info unbound: [30093:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
May 10 22:13:38 OpenMPTCProuter daemon.notice unbound: [12405:0] notice: init module 0: validator
May 10 22:13:38 OpenMPTCProuter daemon.notice unbound: [12405:0] notice: init module 1: iterator
May 10 22:13:38 OpenMPTCProuter daemon.info unbound: [12405:0] info: start of service (unbound 1.17.1).
May 10 22:13:54 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is down
May 10 22:13:54 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity loss
May 10 22:13:57 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is up
May 10 22:13:57 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity 
May 10 22:14:00 OpenMPTCProuter daemon.info unbound: [12405:0] info: generate keytag query _ta-4f66. NULL IN
May 10 22:14:41 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 10 22:14:41 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 10 23:15:32 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 10 23:15:32 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 11 00:16:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 11 00:16:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 11 01:16:53 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 11 01:16:53 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 11 02:17:29 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 11 02:17:29 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 11 03:18:15 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 11 03:18:15 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
May 11 04:18:58 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
May 11 04:18:58 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-snapshot-5.4 r0+16862-1 snap from 09/05
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1031-test
  - OpenMPTCProuter VPS provider: OCI
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-05-11:
Latency is not displayed if this test took too much time.

vempire-ghost, 2024-05-11:
What could be causing the test to take so long? The latency to my VPS is less than 20ms, and running a ping to it during the time when the latency isn't displayed on the OMR, there hasn't been a single instance where it didn't respond.

closed 2024-08-14 by github-actions[bot]