Ysurac/openmptcprouter #3350: Backup to server not working and old ones are not disappearing when I delete them from the VPS - where is all the info stored about the backups?

hhaniel, 2024-05-12:
## Expected Behavior

New backups should get created when I click the button on the router side and when I delete them I would expect them to disappear from the router side and I have rebooted everything as well

## Current Behavior

Can not push backups to VPS and not seeing any errors either - cleared out old stuff from /var/opt/opentcprouter/ on vps but those backups still show on the router side

## Specifications

-OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022
-OpenMPTCProuter VPS version: 0.1030 on Kernel 6.1 (Debian)
-OpenMPTCProuter VPS provider: Fasthosts UK
-OpenMPTCProuter platform: x86_64 on ESX

Ysurac, 2024-05-12:
There is a cache of backups file names on router side in ```uci get openmptcprouter.vps.allbackup```

hhaniel, 2024-05-13:
Thanx - I cleared that and I also cleared the last backup setting which cleaned up the list. - It looks to me like it relies on this list as it only reflects what is on that variable and not what is on the VPS side.

I managed to work out why backups could not be send to the server either: - I had "Save vnstat stats" enabled in the openmptcprouter advanced settings which would create a /etc/vnstat_backup.tar.gz in /etc which over time has gone to big which meant that the curl command line that `/etc/init.d/openmptcprouter backup_send` would use got too long. - This only became visible when I run the command directly after finding that syscal in the html files.

closed 2024-08-16 by github-actions[bot]