Ysurac/openmptcprouter #3352: How to create VPS with VPS script 0.1028 instead of latest version 0.1030

chriskinal, 2024-05-13:
## Expected Behavior
The VPN and VPS script should stay stable throughout the day.

## Current Behavior
The VPN stops and the VPS script is unreachable at random times throughout the day.

## Specifications

  - OpenMPTCProuter version: v0.60 kernel 6.1
  - OpenMPTCProuter VPS version: v0.1030
  - OpenMPTCProuter VPS provider: Amazon Lightsail - 1 GB, 2vCPU, 40 GB
  - OpenMPTCProuter platform: x86_64

My previous system, Router 0.59.1-5.4 and VPS 0.1028, was rock solid through the day. Rarely would I have any issues with connectivity. Both the VPS and router were on scheduled reboots every 2 days.

Since upgrading to router 0.60 and VPS v0.1030, I have experienced random events several times per day where the VPN stops working and the VPS script is unreachable. Both the VPS and router were on scheduled reboots every 2 days. In order to restore service I need to reboot the VPS and the router in that order. Rebooting the router first and then the VPS does not restore service. The problem is particularly noticeable when doing large downloads. Large downloads from a fast server almost always triggers the issue. If a client uses speedtest.net that will also trigger the issue.

I just want to back to what I had before since it gave me no trouble for months at a time.

Ysurac, 2024-05-13:
It's an upgrade or a new install ?
What is the VPN used between router and VPS (in System->OpenMPTCProuter, "Wizard" tab, "Advanced settings" checkbox) ?

chriskinal, 2024-05-13:
It is a clean install on both the router and VPS. It is using Open VPN TCP.

Ysurac, 2024-05-13:
You can try to enable SQM on the connections

Ysurac, 2024-05-13:
Latest VPS version should still work with previous router image if installed with 5.4 kernel.

chriskinal, 2024-05-14:
Are the VPS components different between VPS v0.1028 and v0.1030? Not sure if my issue on the router or VPS side. I suppose I can try v0.1030 with

chriskinal, 2024-05-15:
So far router v0.59.1-5.4 and VPS v0.1030 have been stable and the speed has returned.

chriskinal, 2024-06-03:
I did some more digging on this issue. All testing was done using speetest.net because it allowed me to change between single and multi-connection downloads. The server is using v0.1030 5.4.207-mptcp and the router is using v0.59.1-54. The failures pointed to the server so I started digging into the basics like memory, cpu and disk usage with Htop. What I found was when using a 1G RAM Lightsail server the available RAM was ~940MB and used RAM was ~475MB during idle and light network usage. However, when I started a multi-connection Speetest test the RAM usage climbed to consume all the available RAM and kswapd CPU usage climbed as well but there is no swap file so kswapd was just thrashing. The download speeds were ~200MBps. I switched to a single connection and the used RAM did not climb but the speed dropped to ~60MBps. It looks like is was not using MPTCP due to being a single connection to the server. The multi-connection tests appear to be using MPTCP due to the higher download speed.

I swapped the server out for one with 2GB RAM. The Available RAM is ~1.91GB. During light network usage the the used RAM is about 362MB. Used RAM climbs to ~1.42GB during the multi-connection speed test. During the single connection test, RAM used stayed slightly elevated compared to idle / light usage.

The next step is to rebuild the system using the latest v0.60 OMPTCPRouter on both the server and router. I'll report the findings back to this thread.

closed 2024-09-06 by github-actions[bot]