Ysurac/openmptcprouter #3353: OMR VPS Test version kernel 6.6  and OMR Router - need help for configuration IPV6 ONLY ;-)

vinceducat, 2024-05-13:
with the test version i would like to aggregate between OMR and VPS in ipv6 only

can find how to define (OMR router side ) only 1 ipv6 link between omr and my isp modem (Freebox Pop)
## Current Behavior

need some example or snapshot or best i person doing it ;-)

## Specifications

  - OpenMPTCProuter version: 0.61 kernel 6.6 snapshot
  - OpenMPTCProuter VPS version: 0.1030 test version
  - OpenMPTCProuter VPS provider: Hetzner 
  - OpenMPTCProuter platform: x86_64 (VM on proxmox, Nic in passtrought)  -->

vinceducat, 2024-05-15:
yes but only if i make a mistake OMR router communicate ton OMR vps via ipv4... on the wan link...

my idea is to have this link in ipv6 only .. so how to configure ipv6 only for wan setup...

beacause its not possible in the wizard

vinceducat, 2024-05-15:
so i have eth0 to lan eth1 to wan

wan is connected in direct to Freebox Pop in 2.5gb/s rj45 port ; Freebox in bridge mode..

i have note and confirm with many test (i can detail in private) and have note that ipv4 flow speed (700/300) seems to be limited versus ipv6 (1800/700) perhaps by freebox or isp free on their network.

ipv4 limited limits aggregation because omr uses only ipv4 connection to connect omr and omr vps (am i right ?)
so my goal is to set up OMR router  to send traffic only on ipv6 (beetween omr and omr vps).
so my idea is to define only ipv6 setup for wan and not ipv4 ...

so my config would be have only ipv6 adresse in setup for omr vps and same for wan.

but i want ipv6 and ipv4 from lan would go trought the omr "tunnel" from mor to omr vps in ipv6 and so no limitation in speed du to ipv4 limit from freebox or free network....

(could you image that freebox have qos or other macanism to limit ipv4 traffic and prioritiz ipv6 ?)

Ysurac, 2024-05-15:
If you use snapshot with 6.6 kernel, have both IPv6 and IPv4 configured on WAN and IPv4 and IPv6 address of VPS set, then both IPv6 and IPv4 will be aggregated.
Else you can set only IPv6 on OMR, I don't understand the problem here.
For the Freebox, maybe Free have a network more saturated in IPv4 than IPv6.

vinceducat, 2024-05-15:
@Ysurac i try 6.6 kernel on omr vps on hetzner but seem to be worst in flow rate to done significant compraison and test :-(

i will try to fine a solution do you have vps dealers witch can provide me vps, in fact i have trouble last time in milkywan because they have to do action in case of big failure during ... (its append to me last time ) i wouldnt create more trouble to milky team with some request to restart the vps in cas of failure at the vps side

vinceducat, 2024-05-15:
@Ysurac  redo some test with hetzner but flow rate event in ipv4 or ipv 6 is so slow 40mb/s that a couldnot conlude

closed 2024-08-20 by github-actions[bot]