Ysurac/openmptcprouter #3354: Le vpn ne répond pas

Romain-GUILLEMOT, 2024-05-13:
## Behavior

Depuis une semaine, je rencontre un problème récurrent avec mon installation OpenMPTCProuter qui fonctionnait sans souci depuis plus d'un an. Le message d'erreur "Le VPN ne répond pas" s'affiche de manière persistante, malgré mes tentatives de résolution.
J'ai procédé à la réinstallation complète du serveur VPS ainsi que du routeur Raspberry Pi 4 exécutant OpenMPTCProuter, mais cela n'a pas permis de résoudre le problème. L'erreur persiste et empêche l'établissement de la connexion VPN nécessaire au bon fonctionnement d'OpenMPTCProuter.
Vous trouverez ci-joint une capture d'écran illustrant le message d'erreur, ainsi que les détails de ma configuration actuelle. 

[image]

https://haste.bagou450.com/witaduramo.apache

## Specifications

  - OpenMPTCProuter VPS provider: LibraHost

Ysurac, 2024-05-13:
C'est une configuration mise à jour ?
Si vous allez sur ```https://<VPSIP>:65500/``` avec un navigateur qu'est-ce qu'il affiche ?
Dans Système->OpenMPTCProuter, onglet "wizard", en cochant la case "Paramètres avancés", il faudrait ensuite cocher la case "Forcer la récupération des paramètres". En profiter pour mettre le proxy sur "Shadowsocks-Rust" et mettre le VPN sur "OpenVPN TCP".

Ysurac, 2024-05-14:
Je demandais en fait si c'était une nouvelle installation ou une mise à jour d'une ancienne installation.

closed 2024-08-18 by github-actions[bot]