Ysurac/openmptcprouter #3356: Client install stucks on kmodloader: done loading kernel modules from /etc/modules.d/*

mrroger89, 2024-05-14:
## Expected Behavior
Boot openmptcprouter-v0.60-5.4-r0+16862-170d9e447d-x86-64-generic-squashfs-combined-efi

## Current Behavior
I'm on a n100 platform but, when i am trying to boot the usb drive it stucks on kmodloader: done loading kernel modules from /etc/modules.d/*

## Specifications

  - OpenMPTCProuter version: v0.60-5.4-r0+16862-170d9e447d-x86-64
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider: ServerPlan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-05-15:
Can you try snapshots images ? https://snapshots.openmptcprouter.com/

aquabubble, 2024-08-04:
I am also having the same issue with the latest snapshot on my N100 device. Tried with EFI snapshots for 6.1 and 6.6

Ysurac, 2024-08-07:
You tried to press a key ? As it's should be the end of loading screen.

aquabubble, 2024-08-07:
> You tried to press a key ? As it's should be the end of loading screen.

I'm fairly certain I did. I'll validate and post a screenshot

aquabubble, 2024-08-08:
> You tried to press a key ? As it's should be the end of loading screen.

Seemed I was pressing all key combinations except for Enter, so now I feel very foolish. I'm now in. Thank you for pointing out the obvious!