Ysurac/openmptcprouter #3358: How OMR works?

YB1200, 2024-05-16:
Hello,
I have a question about how OMR works. OMR utilizes MPTCP for aggregation, creating a subflow TCP for each link and primarily sending data through the link with the lowest latency, correct? So, how does this process work exactly? For instance, if you have 10 packets and 3 links, will OMR send 6 packets through the link with the minimum latency and 2 packets through each of the other two links? Is that correct?

kevinh-csalabs, 2024-05-16:
That's a bit simplistic - how it routes is going to be dependent on your settings of whether you optimize for latency or bandwidth etc.

If you choose "Enable TCP Low Latency" then it will try to route based on lowest latency. 

Other things will play a role in routing such as (but not limited to):

- Protocol of data
- Route and bypass configuration
- Multipath TCP path-manager
- Multipath TCP scheduler
- Congestion Control algorithm
- Network monitors (to determine if a specific interface's routes are up or not)

See: https://www.multipath-tcp.org/

YB1200, 2024-05-17:
> For kernel 5.4 MPTCP version used: https://www.multipath-tcp.org/ (used in legacy version of OpenMPTCProuter) For kernel 6.1/6.6 MPTCP version used: https://www.mptcp.dev/ You should have many answer to your questions here.

And for OMR, which policy did you choose to work with it? How can I view that when using it?

YB1200, 2024-05-17:
Yes, that's clear. For the kernel 6.1, how did you configure MPTCP for OMR? Did you choose to prioritize the lowest latency?

closed 2024-08-21 by github-actions[bot]