Ysurac/openmptcprouter #3359: Compile error on macOS (m1)

atyouspace, 2024-05-16:
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of type 'void (void)' [-Wincompatible-function-pointer-types]
__attribute_noreturn__ void (*obstack_alloc_failed_handler) (void)
                              ^
1 error generated.
make[6]: *** [Makefile:2781: lib/obstack.o] Error 1
make[6]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[5]: *** [Makefile:3105: all-recursive] Error 1
make[5]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[4]: *** [Makefile:2321: all] Error 2
make[4]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[3]: *** [Makefile:49: /Volumes/OpenWrt/source/build_dir/host/sed-4.8/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/source/tools/sed'
time: tools/sed/compile#16.14#9.27#49.65
    ERROR: tools/sed failed to build.
make[2]: *** [tools/Makefile:160: tools/sed/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/source'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/source/staging_dir/host/stamp/.tools_compile_yyyyyynnyyyynyyyyyynyynnyyyynyyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/source'
make: *** [/Volumes/OpenWrt/source/include/toplevel.mk:230: world] Error 2

## Specifications

  - OpenMPTCProuter version: 0.6 kernel-5.4

atyouspace, 2024-05-16:
> Maybe you can try to build 0.60 kernel 6.1 that is more up to date.

But I found there is only fullmesh supported in kernel 6.1. is that right?
I want to use other scheduling mechanism with fullmesh together , how to do that?

Ysurac, 2024-05-16:
Other schedulers are available in snapshot using kernel 6.6 (but not yes tested...)

atyouspace, 2024-05-17:
> Maybe you can try to build 0.60 kernel 6.1 that is more up to date.

I build project by `OMR_TARGET="newboard" OMR_FEED_SRC="v0.59.1" OMR_KERNEL="5.4" ./build.sh`.
Is that right? the problem is still there. Are some packages missing?

closed 2024-08-20 by github-actions[bot]