Ysurac/openmptcprouter #3360: The bandwidth of VM aggregation is too low On VirtualBox

kuoxinxin, 2024-05-17:
### Expected Behavior
The bandwidth of the Macbook speed test is the same as that of the Host OS.

### Current Behavior
**Action 1:**
1. On the hostOS, start the OpenMPTCProuter guestOS;
2. Configure 3 Mobile SIM network adapters  on the guestOS as eth1, eth2, and eth3;
3. On the hostOS, use speedtest-cli to test the speed:
[Network]-[MPTCP]-[Bandwidth]-[Download] show the result：
                eth1(master): 47.88Mbps; 
                eth2: 28.46Mpbs; 
                eth3: 29.44Mbps; 
               Total: 102.17Mbps; 

Speedtest-cli download result: 92.05Mpbs.

**Action 2:**
1. On the hostOS, start the OpenMPTCProuter guestOS;
2. Configure 3 Mobile SIM devices network adapters on the guestOS as eth1, eth2, and eth3;
3. Use a Macbook to connect to eth0 of the host through a network cable, and obtain a dynamic IP address through OMR;
4. On your Macbook, use speedtest-cli for speed test:
[Network]-[MPTCP]-[Bandwidth]-[Download] show the result:
                eth1(master): 83.36Mbps; 
                eth2: 21.78Mpbs; 
                eth3: 23.85Mbps; 
                Total: 141.32Mbps;
 
Speedtest download result: 0.76Mpbs.	

### Details:
[image]
[image]

### Host OS speed
[image]

### MacBook speed
[image]

Ysurac, 2024-05-17:
So you seems to aggregate correctly on both case but speed displayed in speedtest seems slower ? Really strange result...
Your MacBook is connected via ethernet or wifi ? Can you try same test with nperf.com if you can find a good server ?

kuoxinxin, 2024-05-17:
**So you seems to aggregate correctly on both case but speed displayed in speedtest seems slower ？**
Yes, it is. Using the same speedtest node, Macbook speedtest is too different from the host test.

**Your MacBook is connected via ethernet or wifi?**
My MacBook connects to omr via ethernet.

**Can you try same test with nperf.com if you can find a good server ?**
I use nperf.com to test bandwidth, but I can't select the Chinese mainland node in the test, I can only use the HK node, and the test result is as follows:
Macbook test: download: 2.487 Mbps
HostOS test: download: 20.42Mbps
remark：the main reason for the low efficiency of the host bandwidth is that the node does not belong to China.

In addition to speedtest-cli, I use [https://www.speedtest.cn/] to test bandwidth. result as follow:
Macbook test download bandwidth: 3.17Mbps
HostOS test download bandwidth: 113.77Mbps

Ysurac, 2024-05-17:
Do you have another device than a Mac to test ? I don't know if the problem is on the mac end or on HostOS when something is connected.

kuoxinxin, 2024-05-17:
I used an Ubuntu 20.04.6 LTS desktop and the results of the tests were the same.Also, I tested the local network bandwidth with iperf3 as follows:
hostOS <--> guestOS bandwidth: (down:1000Mbps; up:1000Mbps);
macbook<-->hostOS bandwidth: (down:936Mbps;up:932Mbps)
**macbook<-->guestos (down:156Kbps;up:4.56Mbps)**

From the above tests, it can be seen that OMR on guestOS may have problems with network transmission. What could be the cause of this problem?

Ysurac, 2024-05-17:
What is your network configuration in VirtualBox ?

kuoxinxin, 2024-05-17:
My Virtualbox network configuration as follows:
### eth0:
[image]

### eth1-3:
usb1,usb2,ubs3 is 3 mobile SIM network adapters.
[image]

closed 2024-08-20 by github-actions[bot]