Ysurac/openmptcprouter #3362: Error ins the console - should I worry?

hairy-dog, 2024-05-17:
No errors to be shown in the console.

## Current Behavior

I installed a new Raspberry pi5 to replace the pi4 I was using for OMR.

While setting it up, I connected a monitor to the hdmi output, and it went as expected for a while, but after a few minutes, the screen fills up with repeated* entries of the same error message:

[ 1234.567890] brcmfmac: brcmf_dump_obss: dup_obss error (-52)

* the number at rhe beginning increments, with the first digits being seconds elapsed since startup, I think

## Possible Solution

Is this safe to ignore, or should I do something to fix it??

## Steps to Reproduce the Problem
This is what I did. I have only done it once.

  1. Clean install of https://download.openmptcprouter.com/release/v0.60-6.1/rpi5/targets/bcm27xx/bcm2712/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2712-rpi-5-squashfs-factory.img.gz
  2. enter VPS and Wan 1, Wan 2 details
  3. Restore backup settings from VPS
  4. Change encryption from chacha20 to AES-256-GCM

## Context (Environment)

I do not know if it affects me at all. OMR seems to be working well.

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 6.1.0-20-cloud-amd64.0.1030 
  - OpenMPTCProuter VPS provider: intovps
  - OpenMPTCProuter platform: RPI5
  - Country: UK

Ysurac, 2024-05-17:
OMR is not designed to have a screen, so screen and sound drivers/patch can be missing.
But in this case, I think error can be related to Wifi, maybe a check on wifi that is failing. Nothing to worry about.

hairy-dog, 2024-05-17:
Thank you. It certainly seems to be working well. Better than with the Pi4, which was good already.

Because there are so many variables, it is difficult to tune OMR for best performance.  Using the Realtime Traffic page seems to help me do that, even though it is generating errors in the console (which I never looked at before).

closed 2024-05-17 by hairy-dog