Ysurac/openmptcprouter #3363: Aggregation glorytun UDP doesn't work  for me..

esquilo444, 2024-05-17:
Good morning,
I can't get the aggregation to work at all in glorytun UDP, with or without a proxy and whatever the Proxy used. the only one that works in vpn is mlvpn, but it is unstable and therefore unusable for my srt video streams.. With v2ray the UDP aggregation also works, but not stable over time either.. I'm a little desperate. The only solution I see is to use one of the proxies for tcp, they all work quite well and manage to get UDP aggregation working by glorytun UDP which has a very stable rtt...
In short, I am open to any suggestions.
thanks 

Jérôme 

## Specifications

  - OpenMPTCProuter version:  Snapshot 6.6
  - OpenMPTCProuter VPS version: kernel test 6.6
  - OpenMPTCProuter VPS provider:ovh
  - OpenMPTCProuter platform: RPI4

Ysurac, 2024-05-17:
With Glorytun UDP, you can check status and path using ```glorytun-udp path``` on the router via SSH

esquilo444, 2024-05-17:
I'm going to try that. I will get back to you...
THANKS
ps udp aggregation via glorytun works for you?

esquilo444, 2024-05-17:
[image]
[image]

esquilo444, 2024-05-17:
i don't no if that can help you ...

esquilo444, 2024-05-17:
[image]

Alison110, 2024-05-18:
somethimes I use glorytun tcp at first then if it was ok, I use glorytun udp.
or 
restart your vps, after 2 minutes go to system ->startup and restart glorytun udp.

esquilo444, 2024-05-18:
I will try it ...
Thanks

esquilo444, 2024-05-19:
unfortunately this does not work.
maybe it's the SRT protocol I'm using to transport the video that's causing the problem...this protocol has its own error corrector..
Do you have a solution to test aggregation via UDP, without me using my software to transport my videos...
this will start to pinpoint the problem.. thanks for your suggestions.

Ysurac, 2024-05-20:
To test aggregation, set proxy to none, so all is using the VPN and you can test aggregation.
There is a know issue with SRT, but I'm a real newbie on using SRT, so not easy to test and would need some times...

esquilo444, 2024-05-20:
Thanks..😅
I am a professional in video and since covid, specialized in streaming, but relatively new to the network part.
I've learned a lot of things in recent years, but I'm failing here, even if I remain convinced that there is a via open mtpcp router...

Ysurac, 2024-05-22:
There is many draft for thinks over quic, but I didn't found anything already developed that I can use for now.

closed 2024-08-26 by github-actions[bot]