Ysurac/openmptcprouter #3367: Some issues and their fix.(linksys wrt3200acm)

Alison110, 2024-05-18:
Hi, first; THANK YOU SO MUCH Ysurac,
but next; I've installed and reinstalled opentcprouter over 30 times and this is some issues and their fix.(linksys wrt3200acm)

1- clients couldn't connect to Wi-Fi because Wi-Fi can't get clients IP, so:
go to: interface⇾ devices⇾ add a bridge device, select wan at bridge ports then, in interface tab: LAN⇾ device and select bridge instead of wan.
Now you can connect to Wi-Fi 
but;

2- bypass only works on LAN or Wi-Fi and now bypass only work on Wi-Fi and, not on LAN (wan) (bug).

3- fastest VPN is glorytun UDP and best proxy is xray Trojan.
But:
Xray couldn't get IP from VPS when use encryption.
xray doesn't work along with v2ray (in my case) so you can set encryption to none for use xray or, use next best proxy which is v2ray with encryption chacha.
if gloryton udf is down; 
restart router and vps both if it's down, go to system ⇾startup, start and restart glorytun UDP again if it's still down use gloryton TCP then back, use gloryton UDP
4- do not update package from software⇾update, especially v2ray!. Stop working Danger!
5- THANK YOU SO MUCH, Ysurac.

## Specifications

  - OpenMPTCProuter version: 6, 6.1, 6.6
  - OpenMPTCProuter VPS version: 6.1&6.6 1030&1031-test
  - OpenMPTCProuter VPS provider: pq.hosting
  - OpenMPTCProuter platform: Linksys WRT3200acm
  - Country: China

QatarMo, 2024-05-19:
my testing with latest 6.1 snapshot try useing
XRay VLESS
AES-256-GCM
Glorytun UDP
no SQM

bypass works for me with this config try it out im on x86 btw

Alison110, 2024-05-19:
> my testing with latest 6.1 snapshot try useing XRay VLESS AES-256-GCM Glorytun UDP no SQM
> 
> bypass works for me with this config try it out im on x86 btw

Bypass works on both interfaces? LAN and WI-FI?

closed 2024-08-22 by github-actions[bot]