Ysurac/openmptcprouter #3369: OPKG custom feed packages missing for aarch64_cortex-a76 (Raspberry Pi 5)

andrewmackrodt, 2024-05-21:
## Expected Behavior
OPKG updates package metadata without errors.

## Current Behavior
`opkg update` is unable to download package metadata for the following lists:
```
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.60-6.1/aarch64_cortex-a76/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.60-6.1/aarch64_cortex-a76/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.60-6.1/aarch64_cortex-a76/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.60-6.1/aarch64_cortex-a76/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://packages.openmptcprouter.com/v0.60-6.1/aarch64_cortex-a76/telephony/Packages.gz, wget returned 8.
```

## Possible Solution
Mirror packages from downloads.openwrt.org (if appropriate). End users can use these lists:
```
    src/gz openwrt_luci https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a76/luci
    src/gz openwrt_packages https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a76/packages
    src/gz openwrt_base https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a76/base
    src/gz openwrt_routing https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a76/routing
    src/gz openwrt_telephony https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a76/telephony

```

## Steps to Reproduce the Problem

  1. Run `opkg update` or click `Update lists...` from `/cgi-bin/luci/admin/system/opkg`

## Context (Environment)
Using the Raspberry Pi 5 SquashFS v0.60-6.1 image:https://download.openmptcprouter.com/release/v0.60-6.1/rpi5/targets/bcm27xx/bcm2712/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2712-rpi-5-squashfs-factory.img.gz

## Specifications

  - OpenMPTCProuter version: v0.60-6.1
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter VPS provider: n/a
  - OpenMPTCProuter platform: RPI5
  - Country: n/a

Ysurac, 2024-05-21:
I added a link to snapshot, should be ok now.

andrewmackrodt, 2024-05-21:
Wow that was super quick, thanks!

closed 2024-05-21 by andrewmackrodt

ayush617, 2024-12-29:
Getting same after update
[image]