Ysurac/openmptcprouter #3371: Aggregation Not working as Expected

Yessine1200, 2024-05-22:
## Expected Behavior
Link Aggregation maximizes the capacity across all connections. (35-45M)

## Current Behavior
It seems like connections are always getting slowed down(for 4G SFR and private 5G)

## Lan1 : 4G ORANGE - 10,7M 
[image]

## Lan2: 4G SFR (very slow)
[image]

## Private 5G - 13,8M
[image]

## Aggregation - 11,4M
[image]
[image]
It seems that the aggregation is working fine, but with limited speed for the two connections (SFR and Private 5G)

## Status page
[image]

## Specifications

  - OpenMPTCProuter version: 6,1
  - OpenMPTCProuter VPS version: 0.1030 6.1 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: BPI-R64

Ysurac, 2024-05-22:
You seems to have a high load average on the BPI-R64.
You can try to add/remove connections, to check if a connection is not limiting aggregated speed.
What is the proxy used ?

Yessine1200, 2024-05-22:
high load average is related to the RAM ?
Shadowsocks Rust for the proxy.

Ysurac, 2024-05-22:
It's related to CPU. Encryption can be CPU intensive.

Yessine1200, 2024-05-29:
Hello Ysurac
Is it possible that one connection could block the other connection from using all its throughput?

Ysurac, 2024-05-29:
Should not, but this can happen

Yessine1200, 2024-05-29:
or can be related to the router ?

Ysurac, 2024-05-29:
yes this can be. BPI-R64 have a slow CPU.

Yessine1200, 2024-05-29:
Because for now, when I aggregate two 4G connections, it works perfectly. I get the correct aggregation, and it's stable.
[image]

But when I try to aggregate the private 5G standalone and 4G connection, I notice that the 4G connection blocks the 5G standalone. However, when I run the command omr-test-speed lan2, I can get the correct throughput of the 5G connection which is 130Mpbs.
[image]

This is why I mentioned that one connection might block the other connection based on the results that I got.

Ysurac, 2024-05-29:
You can try to run ```omr-test-speed lan2``` and ```omr-test-speed the4ginterface``` at the same time, to check the result.

Yessine1200, 2024-05-29:
I first launched omr-test-speed lan2 for the 5G standalone: 
[image]

After 2 minutes, I launched omr-test-speed lan1 for the 4G connection without stopping the speed test for the 5G connection, and I observed that the throughput of the 5G connection decreased as shown below.
[image]

Ysurac, 2024-05-29:
I think you have a congestion point somewhere on the network.

Yessine1200, 2024-05-29:
And you know how can I find this problem ?

Ysurac, 2024-05-29:
by checking where can be the bottleneck on the network, if there is something shared with both network.

Yessine1200, 2024-05-29:
Doesn't the BBRv3 algorithm prevent this from happening?

Ysurac, 2024-05-29:
If it's not possible to have a higher speed due to a bottleneck, no algorithm can do anything.

Yessine1200, 2024-05-31:
> by checking where can be the bottleneck on the network, if there is something shared with both network.

No, there is nothing shared in my network. I am using two different ISPs. It's really weird that I have this issue.

Ysurac, 2024-05-31:
I don't know what your 5G network is using as source ISP/network type. Difficult to say anything about that.
You can test network using 2 phones or devices connected to both network and check the same way if you have same problem or not.

closed 2024-09-06 by github-actions[bot]