Ysurac/openmptcprouter #3372: Use public vps IP on internal firewall

Funstar81, 2024-05-22:
## Expected Behavior
So my testing environment is a vps with public IP 5.75.172.183 and default gateway 172.31.1.1 at Hetzner. My local omr is running in a proxmox vm. They are successfully connected by fibre (TNG) and cable (Vodafone). On my local end, I would like to place a firewall on the LAN side of omr, but use the public IP address of the vps configured at the WAN interface of my firewall.

The idea is that firewall WAN is 5.75.172.183/32 onlink and connected to the LAN interface of omr, which I gave 172.31.1.1/24 aswell (just like Hetzner does). When now a packet comes from internal network through the firewall, it is SNAT to 5.75.172.183 and should be forwarded by omr to 10.255.252.1 as this is the default gateway. Reaching the vps, the packet would simply be forwarded to the internet. So not further NAT or something is needed. I already tried it but tcpdump on the vps told me that omr did a SNAT to 10.255.252.2, which I would want to disable.

Now the other direction, packet comes from the internet to the vps. Here I activated "forward all traffic vom vps to local router" and tcpdump on omr tells me that the packets successfully arrive. Now I would like to add a rule like "when packets come from tun0 and are not for 10.255.252.2, then forward them to LAN interface and do a DNAT to 5.75.172.183. Then those packets would reach my firewall just as they should (don't they?).

So could you please help me with disabling outgoing SNAT and how and where to place that rule for incoming traffic?

## Current Behavior
Outgoing traffic from LAN gets SNAT, incoming traffic from tun0 is not forwarded to LAN interface and not DNAT.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60-61 r0+24843-acf40c022
  - OpenMPTCProuter VPS version: 6.1.0-20-amd64 0.1030
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-05-22:
I'm not sure I really understand what you want.
For incoming packets there is no real problem, you can disable masquerade on VPN zone.
For outgoing packets I don't really understand what you want but you need to know that, by default, TCP is using a proxy, it's a big difference with a VPN.

Ysurac, 2024-05-22:
So you want something like a bridge mode, for me it's not possible.
But you can still try and if you success I can add needed settings for that.

Funstar81, 2024-05-23:
I would like to try it, if you could give me some hints for that:
1) How to disable SNAT for omr => tun0
2) How to enable DNAT for omr => lan

Funstar81, 2024-05-23:
After setting "option masq" to '0' in /etc/config/firewall on my local omr instance, I can see the packets with the unchanged sender ip 5.75.172.183 when I monitor the traffic with tcpdump -i tun0 -n outbound. However, on the vps side of the tunnel, I cannot see those packets with tcptump -i tun0 -n inbound. They get lost somewhere on their way.

closed 2024-08-27 by github-actions[bot]