Ysurac/openmptcprouter #3373: Problem with localization

infosb, 2024-05-23:
Hello, since few day I have a probleme wifi openmptcprouter... I think i have a problem on my vps ...
I'm in france and when i go on firefox on www.google.fr in the bottom left corner i can see "russie" and all answer of my search are in Russian.
I think that my vps have been Hacked ...
I don't know how to find the config on VPS (dns) My public IP is located in Paris.
Have you some ideas ? what can i Verify ?
Thanks for your help

  - OpenMPTCProuter version: OpenMPTCProuter v0.60
  - OpenMPTCProuter VPS version: OpenMPTCProuter v0.60
  - OpenMPTCProuter VPS provider: aruba cloud
  - OpenMPTCProuter platform: x86_64

infosb, 2024-05-23:
Thanks !
But i have 50 pcs in my school.
and i have try with some pc but it don't work and my search results are in russian ! 
But there is a strange thing on google chrome it all ok google.fr and search in french .... on firefox opera brave it's in russian
I don't understand

Ysurac, 2024-05-23:
Language of Google can also be modified by browser.
I think you can force languages by using URL like https://www.google.fr/search?hl=fr&lr=lang_fr (it's maybe the URL chrome is using...)

infosb, 2024-05-23:
Ok but with https://www.google.fr/search?hl=fr&lr=lang_fr there is Russian on the bottom left corner ;) and on youtube many videos are not allow because we are localized in Russian but when i test my public IP i'm in france ...

Ysurac, 2024-05-23:
I have no more idea, maybe google support can help...

Network-Traditions, 2024-06-01:
Your issue is caused by the following:

1. The public IP address attached to your VPS for OpenMPTCProuter (OMR).
2. The GEOip service used by the content provider you are accessing at the time.

Regarding #1, you may be able to get a different IP address for your VPS that is more consistently associated with your location.  We've had this problem and have not been very successful in getting the provider to understand the problem let alone provide an IP address that is properly identified for localization in the various GEOip databases.

Regarding #2, many of the GEOip database service have ways to update localization for public ip addresses.  Here in the USA, Maxmind is one of the predominant providers.  Here is their webpage to make such a correction request: [Submit GeoIP Data Corrections](https://www.maxmind.com/en/geoip-data-correction-request).

When we first launched OMR, we were using IONOS for our VPS.  Initially, our GEOip located us in the USA, but not actually where our VPS data center was located.  After some months, all of a sudden our GEOip located us in Mexico so a great deal of our content was defaulted to the Spanish language.  That's when we researched the GEOip service used by various content providers and contacted the same to request a localization data correction for our IP address.  We did not pursue it to the level of 100% so I'm not sure how extensive that would be.  In the end, we found IONOS quality of service was lacking for OMR for reasons unknown.  We switched to another VPS provider and enjoyed much better OMR reliability, latency and bandwidth.  We suspect latency has the greatest impact on OMR performance.

Finally, we recently upgraded to the latest stable release of OMR/VPS:
OpenMPTCProuter version: 0.60-6.1
OpenMPTCProuter VPS version: 0.1030 6.1.0-21-cloud-arm64
OpenMPTCProuter VPS provider: Oracle Cloud Ampere ARM Instance
OpenMPTCProuter platform: Debian 12 x86_64

Oracle has clean IP addresses for the data center we use, which is also the city where we operate.  On top of the improved performance of this version of OMR/VPS, utilizing a VPS provider as close to our WAN points of presence as possible to reduce latency and utilizing the Ampere ARM offering from Oracle, our OMR is performing extremely well and is a significant improvement over our last VPS provider.  That being said, Oracle's x86 VPS offerings were unusable no matter how robust we configured the VPS.  Our suspicion is the network throughput for the two different offerings is the reason for the dramatically different OMR performance.  Good luck in resolving your localization issues!

infosb, 2024-06-03:
Thanks for all your advices.
I have make a message to google support ... no reponse but few days later it was all ok.
I don't know where the problem was but it works now!

closed 2024-06-03 by infosb