Ysurac/openmptcprouter #3374: Aggregation only works for upload, download is not aggregated

tokon2000, 2024-05-26:
## Expected Behavior

Aggregation of 2x 1000 mbit down / 50 mbit up should give me at least 1600 mbit down / 90 mbit up

## Current Behavior

Speedtest shows me about 800 mbit down and 90 mbit up.
When I test each interface, I get about 900 mbit, but only 800 mbit aggregated. What's wrong?
```
root@OpenMPTCProuter:/bin# ethtool eth0 | grep Speed
	Speed: 1000Mb/s
root@OpenMPTCProuter:/bin# ethtool eth1 | grep Speed
	Speed: 1000Mb/s
root@OpenMPTCProuter:/bin# ethtool eth2 | grep Speed
	Speed: 2500Mb/s
root@OpenMPTCProuter:/bin# omr-test-speed eth0
Select best test server...
host: nbg1-speed.hetzner.com - ping: 18
Best server is https://nbg1-speed.hetzner.com/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 31  9.7G   31 3180M    0     0   110M      0  0:01:30  0:00:28  0:01:02  111M^C
root@OpenMPTCProuter:/bin# omr-test-speed eth1
Select best test server...
host: nbg1-speed.hetzner.com - ping: 17
Best server is https://nbg1-speed.hetzner.com/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 21  9.7G   21 2196M    0     0   109M      0  0:01:31  0:00:20  0:01:11  110M^C
root@OpenMPTCProuter:/bin# omr-test-speed
Select best test server...
host: nbg1-speed.hetzner.com - ping: 18
Best server is https://nbg1-speed.hetzner.com/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 27  9.7G   27 2753M    0     0  88.4M      0  0:01:53  0:00:31  0:01:22 90.6M^C
```
[image]
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 г0+24843-acf40c022
  - OpenMPTCProuter VPS version: 6.1.0-20-amd64 0.1030
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-05-26:
What is the speed limit on the VPS ? Did you try with ```omr-test-speed``` on the VPS ?
There is also ```omr-test-speed-server vps eth0``` that can test speed between eth0 and VPS.
What tis the latency you have between each connection and the VPS (should be visible in status page) ?

tokon2000, 2024-05-26:
OMG, found the problem. I forgot a part of my network that is only connected at 1Gbit.
Now I have 1800 mbit down / 95 mbit up.

closed 2024-05-26 by tokon2000