Ysurac/openmptcprouter #3377: I want install the OPENMPTCPROUTER on a Proliant DL360 Gen10 plus

Gregg03, 2024-05-27:
hello,

I want install the server on my Proliant DL360 Gen10 plus, it's posible ?

If it's posible what image can I use for the installation, the server it's a x86-64 it's use UEFI.

QatarMo, 2024-05-27:
i have it running on a Dell R630 UEFI,,you need to 
1.install the latest Debian 12 then update it 
2.Run the server script then reboot 

i have only 1x 10G eathernet on this R630

Gregg03, 2024-05-28:
Isn't this the command I need to enter on my server to install openmptcprouter?

gunzip openmptcprouter-*.img.gz
dd bs=4M if=openmptcprouter-*.img of=/dev/sdX conv=fsync

Ysurac, 2024-05-28:
For the router part yes. Via an USB key for example.

Ysurac, 2024-05-28:
You can use https://download.openmptcprouter.com/release/v0.60-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz

Gregg03, 2024-05-28:
I will try and I will tell you

Alison110, 2024-05-28:
> Proliant DL360 Gen10 plus

How can I install oopenmptcprouter on (Netgear wndr4300 v1 and mi 4A gigabit) my friends Router's?
They currently have OpenWrt 23.05.
Does anybody know overall steps?

Alison110, 2024-05-28:
> @Alison110 it's not related to this issue. And these devices are not supported, you will need to compile it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

I'm Sorry and Thanks ysurac, Do I need docker composer in debian to compile main.yml file?
( https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml)