Ysurac/openmptcprouter #3378: I want to add a wireless network card to the router

lkyddyy, 2024-05-28:
My router is Raspberry Pi CM4, and there is a MINIPCIE interface on the motherboard. I tried MT7612EN and AR5B22 wireless network cards, but they do not support it. Perhaps because there is no driver, they can run perfectly on X86. Can you recommend a wireless network card that can be used immediately after plugging it in, as I do not know how to compile the system? Thank you

Ysurac, 2024-05-28:
Many drivers can be installed via System->Software menu.

lkyddyy, 2024-05-28:
Thank you very much for your prompt reply. If you can find the driver in the software menu, would it be possible to see the wireless option in the network menu after installation, and then set it up to use

lkyddyy, 2024-06-06:
I'm very sorry to bother you again. I did encounter a problem. I am currently using the MT7921 wireless network card and have installed the driver in the system-software menu. However, the wireless menu still does not appear. I'm not sure where the problem lies. Can you help me，I am using Raspberry Pi cm4
[image]
[image]
[image]

closed 2024-09-10 by github-actions[bot]