Ysurac/openmptcprouter #3382: @Alison110 it's not related to this issue. And these devices are not supported, you will need to compile it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

Alison110, 2024-05-29:
@Alison110 it's not related to this issue. And these devices are not supported, you will need to compile it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

_Originally posted by @Ysurac in https://github.com/Ysurac/openmptcprouter/issues/3377#issuecomment-2135572831_
            
Hello, I compiled kernel 6.6 for xiaomi 4A Gigabit via dear Ysurac instructions
but for kernel 6.1 , there is 48 error messages like tthis:
ERROR: please fix feeds/packages/devel/delve/Makefile - see logs/feeds/packages/devel/delve/dump.txt for details
ERROR: please fix feeds/packages/devel/gitlab-runner/Makefile - see logs/feeds/packages/devel/gitlab-runner/dump.txt for details
.
.
.
ERROR: please fix feeds/openmptcprouter/grpcurl/Makefile - see logs/feeds/openmptcprouter/grpcurl/dump.txt for details.

2 questions:
 1- Does anyone know how to fix this feed errors?
 2- what is the command to build image from compiled file (in bin folder)?

Ysurac, 2024-05-30:
WARNING and ERROR are not important here.
The only problem is the last one: the generated file is far too big... This mean that this rooter have a really small flash available.
This device seems to have 16MB flash and 128MB memory with MT7621 @800Mhz... For me it's far too slow with too small flash and memory to make OpenMPTCProuter running on this.

Alison110, 2024-05-30:
Yest, exactly, the flash is too low. Thank you. Nevertheless, the compilation of Kernel 6.6 was successful, and I am curious what happens after installing mptcprouter on a weak device.
How can I build an image from compiled files (in source/bin)?

closed 2024-05-31 by Alison110