Ysurac/openmptcprouter #3384: OPENMPTCPROUTER can't start

Gregg03, 2024-05-31:
## Expected Behavior
I trying to start my OPENMPTCPROUTER but he can start 

## Current Behavior
I have this error Waiting for root device PARTUUID=Zaf91ce4-288B......
[image]

[image]

## Specifications
  - OpenMPTCProuter version: openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz)
  - OpenMPTCProuter platform: x86_64)

Ysurac, 2024-06-01:
Your device only support UEFI ? No Bios legacy mode ? What is the device ?

Gregg03, 2024-06-01:
I thing he has Bios legacy mode.
It’s a HPE Proliant DL360 Gen10  server

Gregg03, 2024-06-01:
I think

Ysurac, 2024-06-01:
If it has legacy bios, enable this and try the image without efi support.

Gregg03, 2024-06-03:
I tried legacy bios with the good image ans i have that 
[image]

Ysurac, 2024-06-03:
press enter, this may be finished.

Gregg03, 2024-06-03:
when i press enter, I have nothing

Ysurac, 2024-06-03:
Strange, this seems to be the end of the boot, when pressing enter you should get "login:"
Can you try to ping it ?

Gregg03, 2024-06-03:
I can't ping the address 192.168.100.1

closed 2024-09-06 by github-actions[bot]