Ysurac/openmptcprouter #3385: Feasibility of using OMR for WAN Bonding with Limited TCP Ports (80 & 443)

Gitfoe, 2024-06-02:
I'm considering OMR for WAN bonding, but I have a specific scenario that raises a question. One of my WAN connections restricts most internet access, allowing only outgoing connections to TCP ports 80 and 443. Since OMR typically uses ports 65000-65535, is it still possible to use OMR to aggregate both UDP and TCP traffic over this WAN interface? For example, would it be feasible to move Shadowsocks to TCP port 443 and tunnel both UDP and TCP traffic through Shadowsocks?

Ysurac, 2024-06-03:
You can use XRay VLESS Reality proxy, that use port 443,  in System->OpenMPTCProuter, Wizard tab, "advanced settings" checkbox, set XRAY VLESS REality as proxy and disable the VPN, then set in "Advanced settings" tab, the checkbox "V2Ray/XRay UDP".

Gitfoe, 2024-06-05:
Thank you for your quick comment. This works. The only thing that I'm currently having an issue with is DNS. If all the unrestricted WANs are down, DNS lookup requests do not work for the restricted WAN. Outgoing port 53 is also blocked on this WAN (only TCP 80 and 443 are available). This only happens when my end device's DNS is set to 192.168.100.1. DNS works when manually set on the end device, i.e. 8.8.8.8.

Ysurac, 2024-06-11:
Without the VPN, possible that DNS doesn't use the proxy locally... I need to check that. Maybe also possible to use DNS over HTTPS

closed 2024-09-15 by github-actions[bot]