Ysurac/openmptcprouter #3390: OMR can't contact server admin script, OMR-ADMIN keeps restarting, OMR6IN4 doesn't start

Gitfoe, 2024-06-03:
## Expected Behavior
I expect my router to connect to the VPS and communicate via omr-admin and get all its keys and data via the API.

## Current Behavior
I set up my home VPS on a Raspberry Pi, forwarded ports 65000-65535 and 443. On the router, after setting up a single unrestricted WAN and entering the Server Key, it didn't connect. I manually added ShadowSocks, ShadowSocks 2022, Glorytun, A Dead Simple VPN and MLVPN keys. Now the router has connected to the VPS and internet works, but I get `Can't contact Server Admin Script (No token yet available)`.

[image]

The system log repeatedly shows: `Jun  3 20:11:03 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xx.xx.xx.xx, no server API answer on xx.xx.xx.xx)`

Further diagnosis shows omr-admin is probably not running.
```
root@OpenMPTCProuter:~# curl -k https://xx.xx.xx.xx:65500/
curl: (7) Failed to connect to xx.xx.xx.xx port 65500 after 34 ms: Error
```

On the VPS, I noticed omr-admin kept spiking CPU load and then found it kept continuously restarting.
```
Jun 03 22:20:09 DietPi systemd[1]: omr-admin.service: Deactivated successfully.
Jun 03 22:20:09 DietPi systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jun 03 22:20:09 DietPi systemd[1]: omr-admin.service: Consumed 9.001s CPU time.
Jun 03 22:20:09 DietPi systemd[1]: Started omr-admin.service - OMR-Admin.
Jun 03 22:20:22 DietPi systemd[1]: Stopping omr-admin.service - OMR-Admin...
Jun 03 22:20:23 DietPi systemd[1]: omr-admin.service: Deactivated successfully.
Jun 03 22:20:23 DietPi systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jun 03 22:20:23 DietPi systemd[1]: omr-admin.service: Consumed 8.955s CPU time.
Jun 03 22:20:23 DietPi systemd[1]: Started omr-admin.service - OMR-Admin.
Jun 03 22:20:37 DietPi systemd[1]: Stopping omr-admin.service - OMR-Admin...
Jun 03 22:20:38 DietPi systemd[1]: omr-admin.service: Deactivated successfully.
Jun 03 22:20:38 DietPi systemd[1]: Stopped omr-admin.service - OMR-Admin.
Jun 03 22:20:38 DietPi systemd[1]: omr-admin.service: Consumed 9.158s CPU time.
Jun 03 22:20:38 DietPi systemd[1]: Started omr-admin.service - OMR-Admin.
```

I then discovered OMR6IN4 cannot start; this might be the culprit?

Result of `journalctl -xeu omr6in4@user0.service`:
```
Jun 03 22:13:03 DietPi systemd[1]: Starting omr6in4@user0.service - OMR6IN4 on user0...
░░ Subject: A start job for unit omr6in4@user0.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit omr6in4@user0.service has begun execution.
░░
░░ The job identifier is 63580.
Jun 03 22:13:03 DietPi omr-6in4-run[46681]: usage: omr-6in4-run start FILE
Jun 03 22:13:03 DietPi systemd[1]: omr6in4@user0.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit omr6in4@user0.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Jun 03 22:13:03 DietPi systemd[1]: omr6in4@user0.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit omr6in4@user0.service has entered the 'failed' state with result 'exit-code'.
Jun 03 22:13:03 DietPi systemd[1]: Failed to start omr6in4@user0.service - OMR6IN4 on user0.
░░ Subject: A start job for unit omr6in4@user0.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit omr6in4@user0.service has finished with a failure.
░░
░░ The job identifier is 63580 and the job result is failed.
```

My diagnosis ended here. This is a fresh installation of the VPS via the provided script: `wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.1" sh`. Any help is greatly appreciated!

## Specifications
  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 r0+24843-acf40c022e / LuCI Master git-24.006.68576-8debeeb
  - OpenMPTCProuter VPS version: Unsure, I installed it today
  - OpenMPTCProuter VPS provider: Hosted at home on Raspberry Pi 3 Model B+
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-06-05:
You are using as VPS a RPI3 ? This may be the problem... It's really really slow...

Gitfoe, 2024-06-06:
> You are using as VPS a RPI3 ? This may be the problem... It's really really slow...

Yes, as a test I was planning to use what I had on hand. I'm aware it isn't that speedy. Perhaps there's some issue with arm architecture compatibility. But following your comment I've rented a cheap OVH x86_64 Debian 12 VPS which seems to have finished the installation just fine. So I will close this issue for now as I've found a workaround. 😊

closed 2024-06-06 by Gitfoe