Ysurac/openmptcprouter #3391: Any plan to support Ubuntu as VSP host OS other than Debian?

kansonkong, 2024-06-04:
**Is your enhancement proposal related to a problem? Please describe.**

Amzaon AWS doesn't provide Debian as a host OS
**Describe the solution you'd like**

will you please support Ubuntu as VPS other than Debian
**Describe alternatives you've considered**

**Additional context**

vempire-ghost, 2024-06-04:
[image]
There is Debian 12 as a option in AWS.

kansonkong, 2024-06-05:
very embarassed,  amazon aws in China doesn't provide Debian OS

vempire-ghost, 2024-06-05:
> very embarassed, amazon aws in China doesn't provide Debian OS

I use this tutorial to convert Ubuntu to Debian on Oracle Cloud, which also doesn't have Debian as a default option. You can try using it. Although it is specific to Oracle, it might work. Remember that if something goes wrong, it could render the instance unusable, and you would have to recreate it.
https://gist.github.com/ClashTheBunny/8da6f422fc955bfd63ba1b03dd8254c8

kansonkong, 2024-06-08:
thank you, Sir. AWS is different from OCI, I am not lucky to make it.

vempire-ghost, 2024-06-08:
Sad to hear that.

closed 2024-06-20 by kansonkong