Ysurac/openmptcprouter #3393: Aggregation not working

damhau, 2024-06-04:
## Expected Behavior
My two Internet connection bandwidth is aggregated.

## Current Behavior
The download from a client on the LAN is slower than the slowest internet connection.

Here is the bandwith of each compoent:

aggregated (omr-test-speed): 23.2 Mbytes/sec
eth1 (ADSL omr-test-speed eth1): 10.1 Mbytes/sec
eth3 (5G omr-test-speed eth3): 17.6 Mbytes/sec
vps (curl): 58.5 Mbytes/sec
client pc (curl): 4.1 Mbytes/sec

My ADSL connection is 80 Mbits/sec and my 4G connection is approx 150Mbits/sec.

In the status page I've this:

[image]

The system logs are attached here [omr_logs.txt](https://github.com/user-attachments/files/15550517/omr_logs.txt)

I followed the guide in the documentation and I used the install guide here: https://github.com/Ysurac/openmptcprouter/wiki/Router-install#for-x86-and-x86_64 and then the config guide here: https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration

## Specifications

  - OpenMPTCProuter version: [v0.60-6.1 r0+24843-acf40c022e](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: I don't know how to find this version
  - OpenMPTCProuter VPS provider: infomaniak
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-06-04:
If I understand correctly, all is working on the router but your client computer get a slow speed ?
Do you run the router in a VM ? Do you use Wifi ?

damhau, 2024-06-04:
It was wifi. Sorry about that, I'm dumb. I should have tested with my laptop and a cable. ¨
Thanks for the fast answer.

I tested with a cable and I get 250Mbits/sec, it's great. Thank for what you have created.

closed 2024-06-04 by damhau

Ysurac, 2024-06-04:
No problem, thanks for the feedback.

damhau, 2024-06-04:
by the way another nice added feature is that now i've ipv6 connectivity. It is a use case of OMR as well. Nice touch.