Ysurac/openmptcprouter #3395: BPI-R3 Support

YB1200, 2024-06-06:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
I am trying to create an image for the BPI-R3, but so far I have encountered errors. Is it possible to create this image or not?
Maybe, if you want you can guide me to fix this errors.

PS: Even when I tried to create an image for a supported platform, I encountered errors.
**Describe why you are asking for this support?**
I found that the BPI-R3 is better than the BPI-R2 and BPI-R64. It has a good CPU and can achieve effective aggregation.

Ysurac, 2024-06-06:
BPI-R3 should work with develop branch and OMR_KERNEL="6.6" (maybe with 6.1 too) https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

Without the errors, I can't help more.

The CPU seems to be ARM A53 like a Raspberry PI 3. Not so fast...  I hope AES hardware is available, I was not able to find if it's the case or not.

YB1200, 2024-06-06:
I compiled again with kernel 6.6 and got this error:
ERROR: package/feeds/packages/net-snmp failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/bouazizyessine/openmptcprouter/myplatform/6.6/source/include/[toplevel.mk:233](http://toplevel.mk:233/) : world] Erreur 1

I will run it again with `make -j1 V=s` for more details.

YB1200, 2024-06-06:
I got now this error: 
configure: error: Cannot find pcre library. Configure --with-pcre=DIR
make[3]: *** [Makefile:161: /home/bouazizyessine/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_cortex-a53_musl/shadowsocks-libev-3.3.5-410950d87d8cdf8502d8f59a79dc0ff4c7677543/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/bouazizyessine/openmptcprouter/feeds/openmptcprouter/shadowsocks-libev'
time: package/feeds/openmptcprouter/shadowsocks-libev/compile#13.07#1.78#17.94
    ERROR: package/feeds/openmptcprouter/shadowsocks-libev failed to build.
make[2]: *** [package/Makefile:129: package/feeds/openmptcprouter/shadowsocks-libev/compile] Error 1
make[2]: Leaving directory '/home/bouazizyessine/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:123: /home/bouazizyessine/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bouazizyessine/openmptcprouter/myplatform/6.6/source'
make: *** [/home/bouazizyessine/openmptcprouter/myplatform/6.6/source/include/[toplevel.mk:233](http://toplevel.mk:233/) : world] Erreur 2

Ysurac, 2024-06-06:
What is the OS used for compilation ? How much RAM do you have ?

YB1200, 2024-06-06:
I am using Ubuntu 22.04 LTS and I have 32GB of RAM.

Ysurac, 2024-06-06:
Ubuntu is the OS used for compilation via Github Actions (so all the images) (https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml), so you shouldn't have issues. pcre is a dependencies of Shadowsocks-libev, should not make problem, try again.

YB1200, 2024-06-06:
Ok I will try again and back to you. Also I noticed in the logs a lot of warnings, can that also affect the image ?

YB1200, 2024-06-10:
I am getting back to you with another error after waiting 3 hours of running: 

make[3]: *** [Makefile:291: /home/bouazizyessine/openmptcprouter/myplatform/6.6/source/build_dir/target-aarch64_cortex-a53_musl/net-snmp-5.9.4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/bouazizyessine/openmptcprouter/feeds/6.6/packages/net/net-snmp'
time: package/feeds/packages/net-snmp/compile#17.70#4.17#26.12
    ERROR: package/feeds/packages/net-snmp failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/net-snmp/compile] Error 1
make[2]: Leaving directory '/home/bouazizyessine/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:123: /home/bouazizyessine/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bouazizyessine/openmptcprouter/myplatform/6.6/source'
make: *** [/home/bouazizyessine/openmptcprouter/myplatform/6.6/source/include/[toplevel.mk:233](http://toplevel.mk:233/) : world] Erreur 2

Can this error be fixed or I run it again?

YB1200, 2024-06-11:
I still have the same problem with net-snmp. I tried to install some dependencies, but it's still not working.

Ysurac, 2024-06-11:
You can try to disable net-snmp package via ```make menuconfig``` in myplatform/6.6/source directory (you may have to disable openmptcprouter-full package first).

YB1200, 2024-06-12:
For the net-snmp package, I didn't find it via make menuconfig. 
Even after disabling the openmptcprouter-full package, I got the same error.

Ysurac, 2024-06-12:
You can search using the key "/" in menuconfig, I think it's a dependency of luci-app-snmpd, so disable this.

YB1200, 2024-07-30:
> You can try to disable net-snmp package via `make menuconfig` in myplatform/6.6/source directory (you may have to disable openmptcprouter-full package first).

Hello Ysurac,
I’ve just disabled the `openmptcprouter-full` package and successfully generated the image. However, the resulting image is only 5.6MB, which seems unusual. I’d like to understand how I can integrate the full `OMR` package into the image.

Ysurac, 2024-07-30:
You need to clone openmptcprouter-full to another name like yb1200-full and add/remove packages into it. To use it, set OMR_DIST variable when you use build.sh

YB1200, 2024-07-30:
And the image with size 5.6MB is not  weird ?
What do you mean by add/remove packages? Should I check the installed packages, add any that are missing, and remove those that are already installed?

Ysurac, 2024-07-30:
Read what is in the Makefile: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/openmptcprouter-full/Makefile
This is list of packages that need to be installed.

closed 2024-11-02 by github-actions[bot]

esquilo444, 2024-11-02:
Hello YB1200,

have you find a solution for your Banana Pi BPI-R3 ?
I have one and i'm interested... 😋