Ysurac/openmptcprouter #3399: Compile error on linux kernel 6.7

GermanAizek, 2024-06-08:
@Ysurac, hi. Error again when building new kernel.
```
debian@debian-nitro:~/GIT/openmptcprouter/x86_64/6.7/source$ make -j6 V=sc
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_ipset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_nftset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-rtl8812au-ct', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-crypto-lib-blake2s', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/Makefile' has a dependency on 'golang-golang-x-crypto-dev', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/scripts/config'
make[1]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
+ mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/target-x86_64_musl
+ cd /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/target-x86_64_musl/stamp
touch /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/target-x86_64_musl/.prepared
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/nasm'
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/binutils'
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/fortify-headers'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/nasm'
time: toolchain/nasm/compile#0.00#0.00#0.01
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.01#0.02
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/fortify-headers'
time: toolchain/fortify-headers/compile#0.02#0.01#0.04
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/gcc/initial'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.01#0.00#0.02
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/kernel-headers'
rm -rf /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/target-x86_64_musl/root-x86
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
rm -rf /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl/linux-*
xzcat /home/debian/GIT/openmptcprouter/x86_64/6.7/source/dl/linux-github_com_multipath_tcp_mptcp_net_next_git-3d1856db1e4d551bffe5987f57b84b62e245689b.tar.xz | tar -C /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl -xf -
cp -fpR "/home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/files"/. /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl/linux-github_com_multipath_tcp_mptcp_net_next_git-3d1856db1e4d551bffe5987f57b84b62e245689b/
find /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl/linux-github_com_multipath_tcp_mptcp_net_next_git-3d1856db1e4d551bffe5987f57b84b62e245689b/ -name \*.rej -or -name \*.orig | xargs -r rm -f
if [ -d /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/patches ]; then echo "generic patches directory is present. please move your patches to the pending directory" ; exit 1; fi

Applying /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/100-compiler.h-only-include-asm-rwonce.h-for-kernel-code.patch using plaintext: 
patching file include/linux/compiler.h
Hunk #1 succeeded at 199 (offset -4 lines).
Hunk #2 succeeded at 289 (offset 44 lines).

Applying /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/102-MIPS-only-process-negative-stack-offsets-on-stack-tr.patch using plaintext: 
patching file arch/mips/kernel/process.c
Hunk #1 succeeded at 395 (offset 1 line).

Applying /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/103-kbuild-export-SUBARCH.patch using plaintext: 
patching file Makefile
Hunk #1 succeeded at 594 (offset -12 lines).
[... 327 lines cut ...]
patching file drivers/net/phy/realtek.c
Hunk #1 succeeded at 899 (offset 16 lines).
Hunk #2 succeeded at 936 (offset 16 lines).

Applying /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/726-net-phy-realtek-make-sure-paged-read-is-protected-by.patch using plaintext: 
patching file drivers/net/phy/realtek.c
Hunk #1 succeeded at 751 with fuzz 2 (offset 16 lines).

Applying /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/727-net-phy-realtek-use-inline-functions-for-10GbE-adver.patch using plaintext: 
patching file drivers/net/phy/realtek.c
Hunk #1 FAILED at 68.
Hunk #2 FAILED at 669.
Hunk #3 FAILED at 713.
3 out of 3 hunks FAILED -- saving rejects to file drivers/net/phy/realtek.c.rej
Patch failed!  Please fix /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/727-net-phy-realtek-use-inline-functions-for-10GbE-adver.patch!
make[3]: *** [Makefile:123: /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl/linux-github_com_multipath_tcp_mptcp_net_next_git-3d1856db1e4d551bffe5987f57b84b62e245689b/.prepared] Error 1
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#16.42#15.42#22.58
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:93: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make[1]: *** [toolchain/Makefile:89: /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make: *** [/home/debian/GIT/openmptcprouter/x86_64/6.7/source/include/toplevel.mk:233: world] Error 2
```

Ysurac, 2024-06-09:
kerne 6.7 was only a test, it will not be supported. Use kernel 6.6 instead.

GermanAizek, 2024-06-09:
> kerne 6.7 was only a test, it will not be supported. Use kernel 6.6 instead.

6.6 once build error

```
debian@debian-nitro:~/GIT/openmptcprouter/x86_64/6.6/source$ make -j7 V=sc
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_ipset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_nftset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-crypto-lib-blake2s', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-net-rtl8192su', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'kmod-rtl8187', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'luci-i18n-base-en', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/shadowsocks-v2ray-plugin/Makefile' has a dependency on 'golang-golang-x-crypto-dev', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/scripts/config'
make[1]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make[2]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
rm -rf /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl/root-x86
+ mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl
+ cd /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/stamp
touch /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl/.prepared
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make[3]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/target/linux'
make[4]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/target/linux/x86'
rm -f /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/localversion
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/scripts/kconfig.pl  + + /home/debian/GIT/openmptcprouter/x86_64/6.6/source/target/linux/generic/config-6.6 /home/debian/GIT/openmptcprouter/x86_64/6.6/source/target/linux/x86/config-6.6 /home/debian/GIT/openmptcprouter/x86_64/6.6/source/target/linux/x86/64/config-6.6 > /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/debian/GIT/openmptcprouter/x86_64/6.6/source/.config >> /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.target
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/scripts/package-metadata.pl kconfig /home/debian/GIT/openmptcprouter/x86_64/6.6/source/tmp/.packageinfo /home/debian/GIT/openmptcprouter/x86_64/6.6/source/.config 6.6 > /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/.config.override
[... cut ...]
/bin/bash ../libtool  --tag=CC   --mode=compile x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I../include  -I/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/usr/include -I/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/include -I/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/include/fortify  -Wall -W -Wstrict-prototypes  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/sysfsutils-2.1.0=sys

GermanAizek, 2024-06-09:
@Ysurac, main problem
```
curl: (22) The requested URL returned error: 404                              

ERROR: failed to download llvm from ci

    HELP: old builds get deleted after a certain time
    HELP: if trying to compile an old commit of rustc, disable `download-ci-llvm` in config.toml:

    [llvm]
    download-ci-llvm = false
    ```

closed 2024-09-13 by github-actions[bot]