Ysurac/openmptcprouter #3400: "Problème de wan signalés avec une croix rouge" et impossible de contacter le script administrateur

infosb, 2024-06-10:
Hello
Since the last update i have some error 
No token yet avaible 
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030.6.1.0-21 amd64
  - OpenMPTCProuter VPS provider: ARUBA
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-06-10:
The error is more the red cross on all WAN...
You should check status->System log why all interfaces have problems.

Ysurac, 2024-06-10:
It doesn't seems to be able to detect IP/GW on WAN* interfaces.
You also have a very very high load average.
You are running router side on a VM ? What is the hypervisor and why did you not create multiples real interfaces ? (it's seems to be using macvlan here and it's maybe the problem if the VM interfaces is not set in promiscuous mode on the host)

infosb, 2024-06-10:
I'm not on VM It's a intel i3 6100 with 5 lan port.
Before i was on raspeberry pi 4b 
I have change to ethernet port.. it seem to disconnect often ... but we can go to internet

infosb, 2024-06-10:
I have often the message can't connect to script admin

Ysurac, 2024-06-10:
You imported the config from the RPI ? You may have some network config errors if it's the case.
It's more important to have all WAN green than the "can't connect to script admin first.
If, now, all is green and you changed to ethernet port, what do you have again in Status->System log ?

infosb, 2024-06-10:
My wans are not green ....

Jun 10 15:31:53 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Jun 10 15:31:53 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Jun 10 15:31:53 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Jun 10 15:31:53 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 9.9.9.9#53
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jun 10 15:31:53 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jun 10 15:31:54 OpenMPTCProuter authpriv.info dropbear[3991]: Not backgrounding
Jun 10 15:31:54 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 10 15:31:54 OpenMPTCProuter daemon.notice wpa_supplicant[4278]: Successfully initialized wpa_supplicant
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: bonding
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: 8021q
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: macvlan
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: veth
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: bridge
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: Network device
Jun 10 15:31:54 OpenMPTCProuter user.notice : Added device handler type: tunnel
Jun 10 15:31:54 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Jun 10 15:31:54 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is enabled
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is setting up now
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now up
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan5' is enabled
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan5' is setting up now
Jun 10 15:31:55 OpenMPTCProuter daemon.notice netifd: Interface 'wan5' is now up
Jun 10 15:31:55 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='wan5'
Jun 10 15:31:56 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jun 10 15:31:56 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jun 10 15:31:56 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jun 10 15:31:56 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
[... cut ...]
Jun 10 15:32:42 OpenMPTCProuter daemon.info ModemManager[1356]: hotplug: remove network interface SQM_IFB_1d01b: event processed
Jun 10 15:32:42 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4tun0'
Jun 10 15:32:42 OpenMPTCProuter daemon.info ModemManager[1496]: hotplug: add network interface ifb4tun0: event processed
Jun 10 15:32:44 OpenMPTCProuter daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http XXX.XXX.XXX.XXX)
Jun 10 15:32:55 OpenMPTCProuter daemon.info procd: Instance shadowsocks-rust::ss_tunnel.dns s in a crash loop 6 crashes, 0 seconds since last crash

infosb, 2024-06-10:
on test : ping 30 to 45 ms,  download : 50 to 90 mbps, upload 7 to15 mbps

Ysurac, 2024-06-10:
Yes I see still same problems on WANs.
Can you give me the result, via SSH on the router, of ```uci show network``` ?

infosb, 2024-06-11:
Hi !
The result of uci show network

network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.ula_prefix='fd40:2e66:4567::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_stale_loss_cnt='4'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ifname='eth0'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='9999'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.10.200'
network.lan.label='lan'
network.lan.device='eth4'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.addlatency='0'
network.wan1.device='eth0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ipaddr='192.168.3.5'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.3.1'
network.wan1.label='free mobile'
network.wan1.multipath='on'
network.wan1.delegate='0'
network.wan3=interface
network.wan3.proto='static'
network.wan3.ip4table='wan'
network.wan3.defaultroute='0'
network.wan3.metric='7'
network.wan3.peerdns='0'
network.wan3.ipv6='0'
network.wan3.device='eth3'
network.wan3.netmask='255.255.255.0'
network.wan3.ipaddr='192.168.0.2'
[... 1919 lines cut ...]
-ash: -ash:: not found
root@OpenMPTCProuter:~# root@OpenMPTCProuter:~# network.tun0_fw_rule.lookup='150
0'
-ash: root@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# -ash: network.tun0_fw_rule.lookup=1500: not found
-ash: -ash:: not found
root@OpenMPTCProuter:~# root@OpenMPTCProuter:~# network.all_fw_rule=rule
-ash: root@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# -ash: network.all_fw_rule=rule: not found
-ash: -ash:: not found
root@OpenMPTCProuter:~# root@OpenMPTCProuter:~# network.all_fw_rule.priority='1'
-ash: root@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# -ash: network.all_fw_rule.priority=1: not found
-ash: -ash:: not found
root@OpenMPTCProuter:~# root@OpenMPTCProuter:~# network.all_fw_rule.mark='0x4539
'
-ash: root@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# -ash: network.all_fw_rule.mark=0x4539: not found
-ash: -ash:: not found
root@OpenMPTCProuter:~# root@OpenMPTCProuter:~# network.all_fw_rule.lookup='9913
37'
-ash: root@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# -ash: network.all_fw_rule.lookup=991337: not found
-ash: -ash:: not found
root@OpenMPTCProuter:~#

Ysurac, 2024-06-11:
Dirty logs, seems you have pasted in console or something like that.
WAN3 and WAN5 are both using eth3, with WAN5 as macvlan and WAN3 directly. Both must be MACVLAN or shouldn't use same interface.

infosb, 2024-06-11:
Wan 3 and 5 wasn't used.
I have delete them.

infosb, 2024-06-11:
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.ula_prefix='fd40:2e66:4567::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_stale_loss_cnt='4'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ifname='eth0'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='9999'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.10.200'
network.lan.label='lan'
network.lan.device='eth4'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.addlatency='0'
network.wan1.device='eth0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ipaddr='192.168.3.5'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.3.1'
network.wan1.label='free mobile'
network.wan1.multipath='master'
network.wan1.delegate='0'
network.wan4=interface
network.wan4.proto='static'
network.wan4.ip4table='wan'
network.wan4.defaultroute='0'
network.wan4.metric='8'
network.wan4.peerdns='0'
network.wan4.ipv6='0'
network.wan4.device='eth2'
network.wan4.label='sfr'
network.wan4.netmask='255.255.255.0'
network.wan4.ipaddr='192.168.7.10'
network.wan4.gateway='192.168.7.254'
network.wan4.multipath='on'
[... 103 lines cut ...]
network.wan3_dev.name='eth3'
network.eth4_fw_rule=rule
network.eth4_fw_rule.priority='1'
network.eth4_fw_rule.mark='0x45399999'
network.eth4_fw_rule.lookup='9999'
network.eth0_fw_rule=rule
network.eth0_fw_rule.priority='1'
network.eth0_fw_rule.mark='0x45393'
network.eth0_fw_rule.lookup='3'
network.eth2_fw_rule=rule
network.eth2_fw_rule.priority='1'
network.eth2_fw_rule.mark='0x45398'
network.eth2_fw_rule.lookup='8'
network.eth1_fw_rule=rule
network.eth1_fw_rule.priority='1'
network.eth1_fw_rule.mark='0x45394'
network.eth1_fw_rule.lookup='4'
network.tun0_fw_rule=rule
network.tun0_fw_rule.priority='1'
network.tun0_fw_rule.mark='0x45391500'
network.tun0_fw_rule.lookup='1500'
network.all_fw_rule=rule
network.all_fw_rule.priority='1'
network.all_fw_rule.mark='0x4539'
network.all_fw_rule.lookup='991337'

infosb, 2024-06-11:
Sorry for the dirty logs

Ysurac, 2024-06-11:
All is still red in Status page ?

infosb, 2024-06-12:
all my wan are always reds !

Ysurac, 2024-06-12:
What do you have in Status->System log ?
(or you don't want to do a fresh config ?)

infosb, 2024-06-12:
Jun 12 03:59:52 OpenMPTCProuter daemon.err glorytun[12134]: read: Socket not connected
Jun 12 03:59:52 OpenMPTCProuter daemon.info glorytun[12134]: STOPPED tun0
Jun 12 03:59:53 OpenMPTCProuter daemon.err glorytun[12134]: 217.61.97.178.65001: connected
Jun 12 03:59:53 OpenMPTCProuter daemon.err glorytun[12134]: couldn't set socket option `TCP_KEEPCNT'
Jun 12 03:59:53 OpenMPTCProuter daemon.err glorytun[12134]: couldn't set socket option `TCP_KEEPIDLE'
Jun 12 03:59:53 OpenMPTCProuter daemon.err glorytun[12134]: couldn't set socket option `TCP_KEEPINTVL'
Jun 12 03:59:53 OpenMPTCProuter daemon.err glorytun[12134]: couldn't set socket option `TCP_USER_TIMEOUT'
Jun 12 03:59:53 OpenMPTCProuter daemon.info glorytun[12134]: STARTED tun0

I think i will made a fresh config but i must find the best moment for not block internet acces.

Ysurac, 2024-06-12:
Too short logs to say anything (it's only a glorytun VPN restart)

infosb, 2024-06-12:
Jun 12 09:37:06 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Jun 12 09:37:06 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Jun 12 09:37:06 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Jun 12 09:37:06 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Jun 12 09:37:06 OpenMPTCProuter authpriv.info dropbear[3967]: Not backgrounding
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 150
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 9.9.9.9#53
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 8.8.8.8#53
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jun 12 09:37:06 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Jun 12 09:37:07 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Jun 12 09:37:07 OpenMPTCProuter daemon.notice wpa_supplicant[4238]: Successfully initialized wpa_supplicant
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: bonding
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: 8021q
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: macvlan
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: veth
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: bridge
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: Network device
Jun 12 09:37:07 OpenMPTCProuter user.notice : Added device handler type: tunnel
Jun 12 09:37:07 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Jun 12 09:37:07 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Jun 12 09:37:08 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_346f8'
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is setting up now
Jun 12 09:37:08 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is now up
Jun 12 09:37:09 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Jun 12 09:37:09 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Jun 12 09:37:09 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is now up
Jun 12 09:37:10 OpenMPTCProuter daemon.notice netifd: Interface 'wan4' is enabled
[... cut ...]
Jun 12 09:38:05 OpenMPTCProuter daemon.info procd: Instance shadowsocks-rust::ss_tunnel.dns s in a crash loop 6 crashes, 0 seconds since last crash
Jun 12 09:38:23 OpenMPTCProuter daemon.err uhttpd[6562]: [info] luci: accepted login on / for root from 192.168.10.153
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:3319] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:3760] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:4235] N: rops_handle_POLLIN_netlink: DELADDR

Ysurac, 2024-06-12:
Really strange, what is the result of ```ip a``` ?

infosb, 2024-06-12:
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 5acb:231e:1ab8::
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
5: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 5292:e52b:d3ab::
8: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
9: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1400 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:26:55:dc:93:71 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.5/24 brd 192.168.3.255 scope global eth0
       valid_lft forever preferred_lft forever
10: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1400 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:26:55:dc:93:70 brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.10/24 brd 192.168.4.255 scope global eth1
       valid_lft forever preferred_lft forever
11: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1400 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:26:55:dc:93:73 brd ff:ff:ff:ff:ff:ff
    inet 192.168.7.10/24 brd 192.168.7.255 scope global eth2
       valid_lft forever preferred_lft forever
12: eth3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:26:55:dc:93:72 brd ff:ff:ff:ff:ff:ff
13: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 64:00:6a:88:2f:f5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.200/24 brd 192.168.10.255 scope global eth4
       valid_lft forever preferred_lft forever
    inet6 fe80::6600:6aff:fe88:2ff5/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
20: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::78d:a7dc:7ebb:1141/64 scope link stable-privacy proto kernel_ll
       valid_lft forever preferred_lft forever
26: ifb4tun0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 9a:a4:fa:65:f9:ef brd ff:ff:ff:ff:ff:ff

i will test today to do a fresh install if i can

Ysurac, 2024-06-12:
Maybe a problem in omr-tracker configuration, what is the result of ```uci show omr-tracker``` ?

infosb, 2024-06-12:
omr-tracker.defaults=defaults
omr-tracker.defaults.enabled='1'
omr-tracker.defaults.hosts='4.2.2.1' '8.8.8.8' '80.67.169.12' '8.8.4.4' '9.9.9.9' '1.0.0.1' '114.114.115.115' '1.2.4.8' '80.67.169.40' '114.114.114.114' '1.1.1.1'
omr-tracker.defaults.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:fe::fe' '2620:fe::9' '2001:4860:4860::8888' '2001:4860:4860::8844'
omr-tracker.defaults.timeout='2'
omr-tracker.defaults.tries='3'
omr-tracker.defaults.interval='2'
omr-tracker.defaults.interval_tries='1'
omr-tracker.defaults.type='ping'
omr-tracker.defaults.wait_test='0'
omr-tracker.defaults.server_http_test='0'
omr-tracker.defaults.restart_down='0'
omr-tracker.defaults.mail_alert='0'
omr-tracker.proxy=proxy
omr-tracker.proxy.enabled='1'
omr-tracker.proxy.hosts='1.0.0.1' '212.27.48.10' '198.27.92.1' '151.101.129.164' '77.88.55.77' '1.1.1.1' '74.82.42.42' '198.41.212.162'
omr-tracker.proxy.timeout='10'
omr-tracker.proxy.tries='3'
omr-tracker.proxy.wait_test='0'
omr-tracker.proxy.interval_tries='1'
omr-tracker.proxy.interval='5'
omr-tracker.server=server
omr-tracker.server.enabled='1'
omr-tracker.server.tries='3'
omr-tracker.server.timeout='10'
omr-tracker.server.wait_test='0'
omr-tracker.server.interval='5'
omr-tracker.omrvpn=interface
omr-tracker.omrvpn.type='none'
omr-tracker.omrvpn.timeout='10'
omr-tracker.omrvpn.tries='3'
omr-tracker.omrvpn.interval='5'
omr-tracker.omrvpn.mail_alert='0'
omr-tracker.omrvpn.enabled='1'
omr-tracker.omrvpn.server_http_test='1'
omr-tracker.omrvpn.restart_down='0'
omr-tracker.omrvpn.hosts='4.2.2.1' '8.8.8.8'

Ysurac, 2024-06-12:
Can you type: ```uci set omr-tracker.defaults.family='ipv4' && uci commit omr-tracker && /etc/init.d/omr-tracker restart```

infosb, 2024-06-12:
done but all my wans are still in red

infosb, 2024-06-12:
Should i restart

Ysurac, 2024-06-12:
no, the restart of omr-tracker should have done the job...
I would need again system log to check if something changed or not

infosb, 2024-06-12:
Jun 12 09:37:51 OpenMPTCProuter daemon.info ModemManager[31716]: hotplug: add network interface SQM_IFB_b6a77: event processed
Jun 12 09:37:51 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_b6a77'
Jun 12 09:37:51 OpenMPTCProuter daemon.info ModemManager[32219]: hotplug: remove network interface SQM_IFB_b6a77: event processed
Jun 12 09:37:51 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4tun0'
Jun 12 09:37:51 OpenMPTCProuter daemon.info ModemManager[32293]: hotplug: add network interface ifb4tun0: event processed
Jun 12 09:38:05 OpenMPTCProuter daemon.info procd: Instance shadowsocks-rust::ss_tunnel.dns s in a crash loop 6 crashes, 0 seconds since last crash
Jun 12 09:38:23 OpenMPTCProuter daemon.err uhttpd[6562]: [info] luci: accepted login on / for root from 192.168.10.153
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:3319] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:3760] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 09:38:29 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 09:38:29:4235] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 09:44:17 OpenMPTCProuter daemon.info glorytun[12081]: STOPPED tun0
Jun 12 09:44:26 OpenMPTCProuter daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 1.1.1.1, 74.82.42.42, 198.41.212.162)
Jun 12 09:44:27 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jun 12 09:44:43 OpenMPTCProuter daemon.info omr-tracker-ss: Server sss0 (217.61.97.178) seems down, no answer to ping
Jun 12 09:48:43 OpenMPTCProuter daemon.err glorytun[12081]: xxx.xxx.xxx.xxx.65001: connected
Jun 12 09:48:43 OpenMPTCProuter daemon.err glorytun[12081]: couldn't set socket option `TCP_KEEPCNT'
Jun 12 09:48:43 OpenMPTCProuter daemon.err glorytun[12081]: couldn't set socket option `TCP_KEEPIDLE'
Jun 12 09:48:43 OpenMPTCProuter daemon.err glorytun[12081]: couldn't set socket option `TCP_KEEPINTVL'
Jun 12 09:48:43 OpenMPTCProuter daemon.err glorytun[12081]: couldn't set socket option `TCP_USER_TIMEOUT'
Jun 12 09:48:43 OpenMPTCProuter daemon.info glorytun[12081]: STARTED tun0
Jun 12 09:48:49 OpenMPTCProuter daemon.info omr-tracker-ss: Shadowsocks sss0 is up (can contact via http 1.0.0.1)
Jun 12 09:48:49 OpenMPTCProuter daemon.info omr-tracker-ss: Reload Shadowsocks Rust rules
Jun 12 09:48:51 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jun 12 11:19:17 OpenMPTCProuter authpriv.info dropbear[28153]: Child connection from 192.168.10.153:48036
Jun 12 11:19:22 OpenMPTCProuter authpriv.notice dropbear[28153]: Password auth succeeded for 'root' from 192.168.10.153:48036
Jun 12 12:03:12 OpenMPTCProuter user.notice omr-tracker: Launching...
Jun 12 12:03:21 OpenMPTCProuter user.notice omr-tracker: Launched
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan1 (eth0) switched off because wan1 may have ip issues, interface have no IPv4, interface have no IPv4 gateway, interface have no IPv6, interface have no IPv6 gateway
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (eth1) switched off because wan2 may have ip issues, interface have no IPv4, interface have no IPv4 gateway, interface have no IPv6, interface have no IPv6 gateway
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan4 (eth2) switched off because wan4 may have ip issues, interface have no IPv4, interface have no IPv4 gateway, interface have no IPv6, interface have no IPv6 gateway
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 217.61.97.178 dev eth0
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 217.61.97.178 dev eth1
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 217.61.97.178 dev eth2
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: omrvpn down because omrvpn may have ip issues, interface have no IPv4, interface have no IPv4 gateway, interface have no IPv6, interface have no IPv6 gateway
Jun 12 12:03:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Glorytun VPN down, restart it
Jun 12 12:03:21 OpenMPTCProuter daemon.info glorytun[12081]: STOPPED tun0
Jun 12 12:03:21 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 12:03:21:7829] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 12:03:21 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Jun 12 12:03:21 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Jun 12 12:03:21 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Jun 12 12:03:21 OpenMPTCProuter daemon.notice ttyd[28423]: [2024/06/12 12:03:21:7922] N: rops_handle_POLLIN_netlink: DELADDR
Jun 12 12:03:21 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Jun 12 12:03:21 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Jun 12 12:03:21 OpenMPTCProuter daemon.info ModemManager[15271]: hotplug: remove network interface tun0: event processed
Jun 12 12:03:21 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Jun 12 12:03:21 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
[... cut ...]
Jun 12 12:03:24 OpenMPTCProuter daemon.err glorytun[15436]: xxx.xxx.xxx.xxx.65001: connected
Jun 12 12:03:24 OpenMPTCProuter daemon.err glorytun[15436]: couldn't set socket option `TCP_KEEPCNT'
Jun 12 12:03:24 OpenMPTCProuter daemon.err glorytun[15436]: couldn't set socket option `TCP_KEEPIDLE'
Jun 12 12:03:24 OpenMPTCProuter daemon.err glorytun[15436]: couldn't set socket option `TCP_KEEPINTVL'
Jun 12 12:03:24 OpenMPTCProuter daemon.err glorytun[15436]: couldn't set socket option `TCP_USER_TIMEOUT'
Jun 12 12:03:24 OpenMPTCProuter daemon.info glorytun[15436]: STARTED tun0

infosb, 2024-06-12:
I have made a new install my wan are green but i have an error with the time : Big time difference between router and server
my pings are about 500 or more and debit is very low

Ysurac, 2024-06-12:
Set time of the router in System->System and check that your VPS have the correct time in UTC.

infosb, 2024-06-12:
in UTC server and router don't have the same time few secondes

Ysurac, 2024-06-12:
You fixed times ?
I would need status page screenshot and system log.

Ysurac, 2024-06-12:
You have an error in your DHCP configuration. Check what you have in Network->Interfaces, lan, DHCP tab.
You can also give me the result of ```cat /var/etc/dnsmasq.conf.*```

infosb, 2024-06-12:
How can i set times correctly how can i mofify time ?

root@OpenMPTCProuter:~# cat /var/etc/dnsmasq.conf.*
# auto-generated config file from /etc/config/dhcp
conf-file=/etc/dnsmasq.conf
dhcp-authoritative
domain-needed
no-negcache
no-resolv
localise-queries
read-ethers
enable-ubus=dnsmasq
expand-hosts
bind-dynamic
local-service
cache-size=1000
edns-packet-max=1232
domain=lan
local=/lan/
server=127.0.0.1#5353
server=/lan/
server=/use-application-dns.net/
addn-hosts=/tmp/hosts
dhcp-leasefile=/tmp/dhcp.leases
stop-dns-rebind
rebind-localhost-ok
rebind-domain-ok=plex.direct
dhcp-broadcast=tag:needs-broadcast
conf-dir=/tmp/dnsmasq.d
user=dnsmasq
group=dnsmasq

dhcp-ignore-names=tag:dhcp_bogus_hostname
conf-file=/usr/share/dnsmasq/dhcpbogushostname.conf

bogus-priv
conf-file=/usr/share/dnsmasq/rfc6761.conf
dhcp-range=set:lan,19,static,255.255.255.0,12h

nftset=/download.qq.com/steamcontent.com/gs2.ww.prod.dl.playstation.net/dropbox.com/dropboxstatic.com/dropbox-dns.com/log.getdropbox.com/drive.google.com/drive-thirdparty.googleusercontent.com/docs.google.com/docs.googleusercontent.com/gvt1.com/mmg-fna.whatsapp.net/upload.youtube.com/upload.video.google.com/windowsupdate.com/update.microsoft.com/4#inet#fw4#omr_dscp_cs1_4
nftset=/googletagmanager.com/googleusercontent.com/google.com/fbcdn.net/akamaihd.net/whatsapp.net/whatsapp.com/zoom.us/googleapis.com/1e100.net/hwcdn.net/4#inet#fw4#omr_dscp_cs2_4
nftset=/googlevideo.com/nflxvideo.net/s3.ll.dash.row.aiv-cdn.net/d25xi40x97liuc.cloudfront.net/aiv-delivery.net/fbcdn.net/ttvnw.net/vevo.com/audio-fa.scdn.com/deezer.com/sndcdn.com/last.fm/v.redd.it/4#inet#fw4#omr_dscp_cs4_4
nftset=/tv.milkywan.fr/4#inet#fw4#omr_dscp_cs5_4

Ysurac, 2024-06-12:
I have no idea how you can have that in dhcp-range... What do you have in Network->Interfaces, Edit lan, DHCP Server tab ?
As I said for the time, on the router it's in System->System. For the VPS you can install chrony ```apt-get install chrony``` or set the time manually: https://wiki.debian.org/DateTime

infosb, 2024-06-18:
After a new install it seem to be all right.
It's work well about 30ms for the ping, upload about 7 to 12Mbps    and dowload : about 130 à 150 mbps
I have one ADSL one 4g and one 5g 
All items are green ! ;)

infosb, 2024-06-18:
Sometime I have tunnel vpn down but only few secondes

closed 2024-09-22 by github-actions[bot]