Ysurac/openmptcprouter #3401: Starlink + 2 lte connection tunnel down ( under load ) no matter what tunnel i am using

Brazzo978, 2024-06-10:
hi i am trying to understand why on this setup i cant get omr to work stable.

## Current Behavior
Connection 1 : starlink avg 200d 10u
Connection 2 : 4g/lte  avg 80d 50u
Connection 3 : 4g/lte  avg 20d 1u 

i am trying to get around 200d 40u stable troughput , with a feature i absolutly need , remote access via port forwarding from the vps ip, 
but i am not able to figure out why any vpn i select for udp , icmp and port forwarding gets down after some minute of use and then keeps restarting until i stop using my service from outside then the vpn tunnel gets up again , can someone suggest me the best setting for that configuration , now i am using the redundant scheduler and it solves the issue but i get no bandwith aggregation (shadowsock + glorytun tcp ) 

## Specifications

  - OpenMPTCProuter version: 0.59.1 / 0.60-5.4
  - OpenMPTCProuter VPS version: 1030
  - OpenMPTCProuter VPS provider: revolution provider Italy
  - OpenMPTCProuter platform: x64

Ysurac, 2024-06-11:
5.4 based version is a legacy version, so no real changes will be made on this.
Can you try stable 6.1 or snapshot 6.6 ?

Brazzo978, 2024-06-11:
i am trying both versions and with 6.1 i see there are no scheduler sadly , and the tunnel problem is even more present ( openvpn tcp and glorytun tcp/udp)  with 6.6 i see that some scheduler are back but i couldnt find any detail of what they do online , dindt do extensive testing on 6.6 yet because i lack knowledge on the new mptcp upstream version 
[image]
 could you explain what these scheduler do ?

Ysurac, 2024-06-11:
I will add a help text asap.
bpf_burst => same as the default scheduler
bpf_red => sends all packets redundantly on all available subflows
bpf_first => always picks the first subflow to send data
bpf_rr => always picks the next available subflow to send data (round-robin)

Brazzo978, 2024-06-11:
bpf_red  is the substitute to the old redundant ?  amazing gona try it out

Brazzo978, 2024-06-26:
@Ysurac  i have a question but i didn't want to open another issue ,  the wireguard client that the vps creates should be used to access the omr local lan right ?  if so can i know what is used to access lan from the vps ( proxy tunnel ? ) and also if the user changes the Lan subnet , editing the allowed ips from the wireguard config would the tunnel still work ?  thanks

Ysurac, 2024-06-26:
To access the LAN from the VPS, the VPN set on the router is used (can't be wireguard that is used only for MPTCP over VPN workaround).
LAN IPs should be updated via the VPS API using openmptcprouter-vps script on the router.

Brazzo978, 2024-06-26:
so the wireguard confing we find in the vps after the setup should be able to reach the lan of the router if the vpn is up no matter what vpn right ?

Ysurac, 2024-06-26:
Yes, should be able to ping by default.

Brazzo978, 2024-06-28:
hi @Ysurac  i can ping using the tunnel but i cant access the gui of omr , do i need to create a rule to allow traffic ? thanks

Ysurac, 2024-06-28:
You need to allow access from the VPN, in Network->Firewall, at the bottom of the screen set input in VPN line to accept.

Brazzo978, 2024-06-28:
[image]
 i had to edit like that for the whole lan to be reachable using the wg config , leaving the screen here as a documentation on how to do it .

Brazzo978, 2024-06-28:
another question @Ysurac  using v2ray / xray for port forwarding , why is possible to open only 1 port for each rule , and how did you build port forwarding over proxy ?  just curious

closed 2024-10-01 by github-actions[bot]