Ysurac/openmptcprouter #3402: xray.service: Failed

Martronic-SA, 2024-06-12:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 0.60-6.6 
  - OpenMPTCProuter VPS version: Linux  6.1.0-21-amd64 
  - OpenMPTCProuter VPS provider: ovh
  - OpenMPTCProuter platform: x86_64 

Jun 12 11:57:14 de1 OMR-Service[2513829]: ss-server not detected, restart XRay
Jun 12 11:57:14 de1 systemd[1]: Started xray.service - XRay Service.
Jun 12 11:57:14 de1 xray[2513831]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.5 linux/amd64)
Jun 12 11:57:14 de1 xray[2513831]: A unified platform for anti-censorship.
Jun 12 11:57:14 de1 xray[2513831]: 2024/06/12 11:57:14 [Info] infra/conf/serial: Reading config: /etc/xray/xray-server.json
Jun 12 11:57:14 de1 xray[2513831]: Failed to start: main: failed to load config files: [/etc/xray/xray-server.json] > infra/conf: shadowsocks 2022 (multi-user): only blake3-aes-*-gcm methods are supported
Jun 12 11:57:14 de1 systemd[1]: xray.service: Main process exited, code=exited, status=23/n/a
Jun 12 11:57:14 de1 systemd[1]: xray.service: Failed with result 'exit-code'.
Jun 12 11:57:18 de1 ss-server[838822]: getpeername: Transport endpoint is not connected
Jun 12 11:57:23 de1 ss-server[838822]: getpeername: Transport endpoint is not connected
Jun 12 11:57:27 de1 OMR-Service[2514601]: ss-server not detected, restart XRay
Jun 12 11:57:27 de1 systemd[1]: Started xray.service - XRay Service.
Jun 12 11:57:27 de1 xray[2514603]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.5 linux/amd64)
Jun 12 11:57:27 de1 xray[2514603]: A unified platform for anti-censorship.
Jun 12 11:57:27 de1 xray[2514603]: 2024/06/12 11:57:27 [Info] infra/conf/serial: Reading config: /etc/xray/xray-server.json
Jun 12 11:57:27 de1 xray[2514603]: Failed to start: main: failed to load config files: [/etc/xray/xray-server.json] > infra/conf: shadowsocks 2022 (multi-user): only blake3-aes-*-gcm methods are supported
Jun 12 11:57:27 de1 systemd[1]: xray.service: Main process exited, code=exited, status=23/n/a
Jun 12 11:57:27 de1 systemd[1]: xray.service: Failed with result 'exit-code'.
Jun 12 11:57:28 de1 ss-server[838822]: getpeername: Transport endpoint is not connected
Jun 12 11:57:33 de1 ss-server[838822]: getpeername: Transport endpoint is not connected

Ysurac, 2024-06-12:
What is the result of ```uci show xray``` ?

Martronic-SA, 2024-06-12:
I don't understand the question, is this a command to execute?

uci show xray
-bash: uci: command not found

Ysurac, 2024-06-12:
hum sorry it's on VPS side.
I would need the file ```/etc/xray/xray-server.json``` (remove the password/id)

Martronic-SA, 2024-06-13:
Voici:
{
    "log": {
        "loglevel": "error",
        "error": "/tmp/v2rayError.log"
    },
    "transport": {
        "tcpSettings": {},
        "wsSettings": {},
        "kcpSettings": {
            "mtu": 1460,
            "tti": 10,
            "uplinkCapacity": 100,
            "downlinkCapacity": 100,
            "congestion": false,
            "readBufferSize": 8,
            "writeBufferSize": 8
        }
    },
    "inbounds": [
        {
            "tag": "omrin-tunnel",
            "port": 65248,
            "protocol": "vless",
            "settings": {
                "decryption": "none",
                "clients": [
                    {
                        "id": "XXXXX",
                        "level": 0,
                        "alterId": 0,
                        "email": "openmptcprouter"
                    }
                ]
            },
            "streamSettings": {
                "sockopt": {
                    "tcpMptcp": true,
                    "mark": 0
                },
                "network": "tcp",
                "security": "tls",
                "tlsSettings": {
                    "certificates": [
                        {
                            "certificateFile": "/etc/openvpn/ca/pki/issued/server.crt",
                            "keyFile": "/etc/openvpn/ca/pki/private/server.key"
                        }
                    ]
                }
            }
        },
        {
            "tag": "omrin-vmess-tunnel",
            "port": 65250,
            "protocol": "vmess",
            "settings": {
                "decryption": "none",
                "clients": [
                    {
                        "id": "XXXXX",
                        "level": 0,
                        "alterId": 0,
                        "email": "openmptcprouter"
                    }
                ]
            },
            "streamSettings": {
                "sockopt": {
                    "tcpMptcp": true,
[... 150 lines cut ...]
    "api": {
        "tag": "api",
        "services": [
            "HandlerService",
            "LoggerService",
            "StatsService"
        ]
    },
    "policy": {
        "levels": {
            "0": {
                "uplinkOnly": 0,
                "downlinkOnly": 0,
                "bufferSize": 512,
                "connIdle": 2400,
                "statsUserUplink": true,
                "statsUserDownlink": true
            }
        },
        "system": {
            "statsInboundUplink": true,
            "statsInboundDownlink": true
        }
    }
}

Ysurac, 2024-06-13:
It's fixed in latest snapshot: chacha20 encryption is not supported for XRay Shadowsocks proxy so AES-256-GCM is used in all cases.

Martronic-SA, 2024-06-13:
Thank you very much the error is solved, now I have the error: getpeername: Transport endpoint is not connected
here is the configuration:
{
"server": "0.0.0.0",
"port_key": {
"65101": "XXX"
},
"local_port": 1081,
"mode": "tcp_and_udp",
"timeout": 1000,
"method": "aes-256-gcm",
"verbose": 0,
"ipv6_first": true,
"prefer_ipv6": false,
"fast_open": true,
"no_delay": true,
"reuse_port": true,
"mptcp": true,
"ebpf": 0,
"acl": "/etc/shadowsocks-libev/local.acl"
}

Ysurac, 2024-06-13:
You have this error on router side ? The error mean that some website can't be reached via the VPS.

Martronic-SA, 2024-06-13:
no on the vps, I am trying to resolve a connection problem to the vpn because on the router at regular intervals the router cannot reach the vpn

Ysurac, 2024-06-13:
As you are on kernel 6.6, set the proxy to Shadowsocks-Rust in System->OpenMPTCProuter, Wizard tab, "Advanced settings" checkbox.
Should be better than Shadowsocks-libev.
Your VPN is UDP or TCP ?

Martronic-SA, 2024-06-13:
VPN par défaut : Glorytun TCP et MPTCP sur VPN WireGuard

Martronic-SA, 2024-06-13:
I have been using OpenMPTCProuter for a year with version 0.59.1-5.4 and its default configuration, which gave me complete satisfaction. I decided to update to version 0.60-6.6 and eventually reinstalled both the router and the VPS, which is a dedicated server.

Since making changes to OpenVPN TCP, it seems there are no more disconnections.

Donating is normal when there is a project we use regularly, and especially with someone who answers questions, we will support this project regularly.

A big thank you for your project and your help.

closed 2024-09-16 by github-actions[bot]