Ysurac/openmptcprouter #3404: Disconnect every Few Seconds

nicolaeser, 2024-06-13:
## Expected Behavior
Connection should not be disconnected every few seconds.

## Current Behavior
Hey,

I am using OpenMPTCRouter and that is based on OpenWRT afaik!
So my Problem is, that the Router is disappearing every few seconds and is losing connection.
For me i am currently using a RPI 3 with LAN, but a friend also uses some newer Pi with USB Modems and having disconnect issues aswell.

Would appreciate your answers and ideas, what we might have to change or if this is some known bug/problem currently!

Best Regards

## Specifications

  - OpenMPTCProuter version: v0.6
  - OpenMPTCProuter VPS version: My Console shows: < OpenMPTCProuter VPS 0.1030 >
  - OpenMPTCProuter VPS provider: NexoSystems
  - OpenMPTCProuter platform: RPI3 (my one), RPI5 (my friends one)

Ysurac, 2024-06-13:
I would need Status->System log

nicolaeser, 2024-06-13:
@Ysurac Hey Thanks for the fast reply! I requested my friend to get me the system status right now. 

And here it is! Alternatively I could offer you that you write me on discord (@ipexa)

[message.txt](https://github.com/user-attachments/files/15827313/message.txt)

Ysurac, 2024-06-13:
Difficult to be sure as the log have only 2 minutes since boot but it seems that interface are really not stables.
This seems to be USB dongles, do you use a powered USB hub ?

nicolaeser, 2024-06-13:
I forwarded the question to my friend with the devices and getting back to you asap!

nicolaeser, 2024-06-13:
If I understand that right you may think that it is an power issue?

nicolaeser, 2024-06-13:
huawei e3372 directly connected to the pi

Ysurac, 2024-06-14:
Yes, I think it's a power issue. It's often the case when USB is used with RPI.

nicolaeser, 2024-06-14:
Any Idea how you can solve that? Or any other device you can recommend for travelling with backpack and openmptc

Ysurac, 2024-06-15:
Use a powered USB hub or check that you give enough power to the RPI if you are not using the official power supply.

stwr, 2024-06-15:
I agree, likely a power issue. RPi’s generally only supply up to 1.2A shared over **all** 4 usb ports. With 1 or 2 modems connected it is often already enough to cause issues.

As Ysurac says, A powered usb hub (for portability you can get one that supplies power via an additional USB) will do the trick. And make sure you supply your Pi with ample power

closed 2024-09-19 by github-actions[bot]